IOsetting/py32f0-template #1: LL /HAL documentation

prosper00, 2023-02-11:
First: thanks for your work on this, this is great!

Second: do you know if there exists any documentation for the peripherals libraries? They look like clones of the ST libraries, but I'm not sure what differences there are, or if I can / should just use the ST docs

IOsetting, 2023-02-11:
Hi Prosper00, 

So far I don't see a full documentation of peripherals libraries, only the user reference manual and some application notes on seperate peripherals, which can be found in the "PY-MCU资料-20221117.rar" download from https://www.puyasemi.com/cpzx3/info_267_aid_242_kid_235.html.

The file names (and content) are in Chinese, these are translated by machine:
```
├── Application Note
│ ├ ── an1001 _ py32f030 _ py32f003 _ py32f002a series _ use LPTIMER to wake up microcontroller from low power consumption mode. pdf
│ ├ ── an1011 _ py32f030 _ py32f003 _ py32f002a series _ADC application notes. pdf
│ ├ ── an1012 _ py32f030 _ py32f003 _ py32f002a series _GPIO application guide. pdf
│ ├ ── an1013 _ py32f030 _ py32f003 _ py32f002a series _USART hardware automatic baud rate detection. pdf
│ ├ ── an1014 _ py32f030 _ py32f003 _ py32f002a series _ matters needing attention in the application of _SPI. pdf
│ ├ ── an1015 _ py32f030 _ py32f003 _ py32f002a series _I2C application notes. pdf
│ ├ ── an1017 _ py32f030 _ py32f003 _ py32f002a series _ using CRC calculation unit for data verification. pdf
│ ├ ── an1018 _ py32f030 _ py32f003 _ py32f002a series _ How to use RCC reset and clock control module ... pdf
│ ├── AN1021_PY32F030_PY32F003 series _ using RTC real-time clock module. pdf
│ ├ ── an1022 _ py32f030 _ py32f003 _ py32f002a series _ how to use WDG watchdog module. pdf
│ ├── AN1024_PY32F030_PY32F003 series _ How to use LED controller module. pdf
│ ├ ── an1027 _ py32f030 _ py32f003 _ py32f002a series _ Overview of low power consumption characteristics. pdf
│ ├── AN1029_PY32F030_PY32F003 series _ internal HSI calibration application notes. pdf
│ ├── ANSA-1001 PY32F030 series application guide V1.2.pdf
│ ├── ANSA-1002 PY32F030 series interrupt vector mapping application guide V1.0.pdf
│ └── PY32F030_PY32F003 FAQ V1.0.pdf
```

The library structure and naming are similar to those of ST's, but as I know these parts are not compatible, even the pinouts are different. so it might not work to use ST docs.

prosper00, 2023-02-11:
Oh, excellent. And the source files have doxygen comment blocks too.... I'll see about getting a doxyfile set up, some auto generated documentation produced

Witawat, 2023-03-19:
how to develop  on software name ?