IOsetting/py32f0-template #2: Target LIB_FLAGS = PY32F002Ax5 tries to configure USART2 (which doesn't exist on this device)

prosper00, 2023-02-27:
I'm trying to compile Examples/LL/GPIO/LED_Toggle - I copied all the source files from there to User, and edited the Makefile to refer to the PY32F002Ax5 device (linker scripts, pyocd target and libraries). I also edited the source file to toggle porta 5.

When I leave the library flags set to the F030x device, it compiles and flashes fine, and the LED blinks. However, when I set the lib_flags to the correct device, I get the following:

$ make -j

  CC	User/main.c
  CC	User/py32f0xx_it.c
  CC	Libraries/CMSIS/Device/PY32F0xx/Source/system_py32f0xx.c
  CC	Libraries/PY32F0xx_LL_Driver/Src/py32f0xx_ll_pwr.c
  CC	Libraries/PY32F0xx_LL_Driver/Src/py32f0xx_ll_flash.c
  CC	Libraries/PY32F0xx_LL_Driver/Src/py32f0xx_ll_i2c.c
  CC	Libraries/PY32F0xx_LL_Driver/Src/py32f0xx_ll_comp.c
  CC	Libraries/PY32F0xx_LL_Driver/Src/py32f0xx_ll_led.c
  CC	Libraries/PY32F0xx_LL_Driver/Src/py32f0xx_ll_rtc.c
  CC	Libraries/PY32F0xx_LL_Driver/Src/py32f0xx_ll_gpio.c
  CC	Libraries/PY32F0xx_LL_Driver/Src/py32f0xx_ll_crc.c
  CC	Libraries/PY32F0xx_LL_Driver/Src/py32f0xx_ll_spi.c
  CC	Libraries/PY32F0xx_LL_Driver/Src/py32f0xx_ll_lptim.c
  CC	Libraries/PY32F0xx_LL_Driver/Src/py32f0xx_ll_dma.c
  CC	Libraries/PY32F0xx_LL_Driver/Src/py32f0xx_ll_usart.c
  CC	Libraries/PY32F0xx_LL_Driver/Src/py32f0xx_ll_adc.c
  CC	Libraries/PY32F0xx_LL_Driver/Src/py32f0xx_ll_rcc.c
  CC	Libraries/PY32F0xx_LL_Driver/Src/py32f0xx_ll_exti.c
  CC	Libraries/PY32F0xx_LL_Driver/Src/py32f0xx_ll_utils.c
  CC	Libraries/PY32F0xx_LL_Driver/Src/py32f0xx_ll_tim.c
  CC	Libraries/BSP_LL/Src/py32f0xx_bsp_led.c
  CC	Libraries/BSP_LL/Src/py32f0xx_bsp_printf.c
  CC	Libraries/BSP_LL/Src/py32f0xx_bsp_button.c 
  AS	Libraries/CMSIS/Device/PY32F0xx/Source/gcc/startup_py32f002.s
  CC	Libraries/BSP_LL/Src/py32f0xx_bsp_clock.c
In file included from Libraries/BSP_LL/Src/py32f0xx_bsp_printf.c:4:
Libraries/BSP_LL/Src/py32f0xx_bsp_printf.c: In function 'BSP_UART_TxChar':
./Libraries/BSP_LL/Inc/py32f0xx_bsp_printf.h:34:49: error: 'USART2' undeclared (first use in this function); did you mean 'USART1'?
   34 | #define DEBUG_USART                             USART2
      |                                                 ^~~~~~
Libraries/BSP_LL/Src/py32f0xx_bsp_printf.c:10:28: note: in expansion of macro 'DEBUG_USART'
   10 |     LL_USART_TransmitData8(DEBUG_USART, ch);
      |                            ^~~~~~~~~~~
./Libraries/BSP_LL/Inc/py32f0xx_bsp_printf.h:34:49: note: each undeclared identifier is reported only once for each function it appears in
   34 | #define DEBUG_USART                             USART2
      |                                                 ^~~~~~
Libraries/BSP_LL/Src/py32f0xx_bsp_printf.c:10:28: note: in expansion of macro 'DEBUG_USART'
   10 |     LL_USART_TransmitData8(DEBUG_USART, ch);
      |                            ^~~~~~~~~~~
Libraries/BSP_LL/Src/py32f0xx_bsp_printf.c: In function 'BSP_USART_Config':
./Libraries/BSP_LL/Inc/py32f0xx_bsp_printf.h:35:74: error: 'LL_APB1_GRP1_PERIPH_USART2' undeclared (first use in this function); did you mean 'LL_APB1_GRP2_PERIPH_USART1'?
   35 | #define DEBUG_USART_CLK_ENABLE()                LL_APB1_GRP1_EnableClock(LL_APB1_GRP1_PERIPH_USART2)
      |                                                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~
Libraries/BSP_LL/Src/py32f0xx_bsp_printf.c:53:5: note: in expansion of macro 'DEBUG_USART_CLK_ENABLE'
   53 |     DEBUG_USART_CLK_ENABLE();
      |     ^~~~~~~~~~~~~~~~~~~~~~
./Libraries/BSP_LL/Inc/py32f0xx_bsp_printf.h:34:49: error: 'USART2' undeclared (first use in this function); did you mean 'USART1'?
   34 | #define DEBUG_USART                             USART2
      |                                                 ^~~~~~
Libraries/BSP_LL/Src/py32f0xx_bsp_printf.c:56:26: note: in expansion of macro 'DEBUG_USART'
   56 |     LL_USART_SetBaudRate(DEBUG_USART, SystemCoreClock, LL_USART_OVERSAMPLING_16, baudRate);
      |                          ^~~~~~~~~~~
Libraries/BSP_LL/Src/py32f0xx_bsp_printf.c: In function '__io_putchar':
./Libraries/BSP_LL/Inc/py32f0xx_bsp_printf.h:34:49: error: 'USART2' undeclared (first use in this function); did you mean 'USART1'?
   34 | #define DEBUG_USART                             USART2
      |                                                 ^~~~~~
Libraries/BSP_LL/Src/py32f0xx_bsp_printf.c:106:28: note: in expansion of macro 'DEBUG_USART'
  106 |     LL_USART_TransmitData8(DEBUG_USART, ch);
      |                            ^~~~~~~~~~~
Libraries/BSP_LL/Src/py32f0xx_bsp_printf.c: In function '__io_getchar':
./Libraries/BSP_LL/Inc/py32f0xx_bsp_printf.h:34:49: error: 'USART2' undeclared (first use in this function); did you mean 'USART1'?
   34 | #define DEBUG_USART                             USART2
      |                                                 ^~~~~~
Libraries/BSP_LL/Src/py32f0xx_bsp_printf.c:116:40: note: in expansion of macro 'DEBUG_USART'
  116 |     while (!LL_USART_IsActiveFlag_RXNE(DEBUG_USART));
      |                                        ^~~~~~~~~~~
make: *** [rules.mk:80: Build/Libraries/BSP_LL/Src/py32f0xx_bsp_printf.o] Error 1
make: *** Waiting for unfinished jobs....
Libraries/BSP_LL/Src/py32f0xx_bsp_clock.c: In function 'BSP_RCC_HSI_PLL48MConfig':
Libraries/BSP_LL/Src/py32f0xx_bsp_clock.c:14:3: warning: implicit declaration of function 'LL_PLL_ConfigSystemClock_HSI' [-Wimplicit-function-declaration]
   14 |   LL_PLL_ConfigSystemClock_HSI(&UTILS_ClkInitStruct);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
Libraries/BSP_LL/Src/py32f0xx_bsp_clock.c: In function 'BSP_RCC_HSE_PLLConfig':
Libraries/BSP_LL/Src/py32f0xx_bsp_clock.c:65:3: warning: implicit declaration of function 'LL_PLL_ConfigSystemClock_HSE' [-Wimplicit-function-declaration]
   65 |   LL_PLL_ConfigSystemClock_HSE(HSE_VALUE, LL_UTILS_HSEBYPASS_OFF, &UTILS_ClkInitStruct);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~

prosper00, 2023-02-27:
Also, it does work if I go through the BSP_LL/Inc/py32f0xx_bsp_printf.h file and find/replace all occurrences of USART2 to USART1, as well as change this line to 'LL_APB1_GRP2_PERIPH_USART1' as suggested by the build. [The 'LL_PLL_ConfigSystemClock_HSI' and 'LL_PLL_ConfigSystemClock_HSE' still come up though]

./Libraries/BSP_LL/Inc/py32f0xx_bsp_printf.h:35:74: error: 'LL_APB1_GRP1_PERIPH_USART1' undeclared (first use in this function); did you mean 'LL_APB1_GRP2_PERIPH_USART1'?
   35 | #define DEBUG_USART_CLK_ENABLE()                LL_APB1_GRP1_EnableClock(LL_APB1_GRP1_PERIPH_USART1)
      |                                                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~

IOsetting, 2023-02-28:
Hi prosper00,

These examples and bsp functions were written and tested with py32f030x4 and py32f030f8 flags. I didn't test them with py32f002a flag because all 002a, 003 and 030 are the same.

You can use flag py32f030x4 on py32f002a chips, everything on 030 also works on 002a.

IOsetting, 2023-03-12:
Because all parts have USART1, I have changed the bsp debug port to USART1. 
The bsp printf should work on PY32F002Ax5 now.

prosper00, 2023-03-13:
Lovely. Works like a charm now; no errors. 'blink' example at least now functions as I'd expect it to on my chip

prosper00, 2023-04-03:
I found some more issues with 'LIB_FLAGS = PY32F002Ax5'
```
$ make 
  CC	User/main.c
In file included from ./Libraries/PY32F0xx_LL_Driver/Inc/py32f0xx_ll_adc.h:32,
                 from User/main.h:8,
                 from User/main.c:5:
./Libraries/PY32F0xx_LL_Driver/Inc/py32f0xx_ll_rcc.h: In function 'LL_RCC_HSE_EnableCSS':
./Libraries/PY32F0xx_LL_Driver/Inc/py32f0xx_ll_rcc.h:584:20: error: 'RCC_CR_CSSON' undeclared (first use in this function); did you mean 'RCC_CR_HSION'?
  584 |   SET_BIT(RCC->CR, RCC_CR_CSSON);
      |                    ^~~~~~~~~~~~
./Libraries/CMSIS/Device/PY32F0xx/Include/py32f0xx.h:175:42: note: in definition of macro 'SET_BIT'
  175 | #define SET_BIT(REG, BIT)     ((REG) |= (BIT))
      |                                          ^~~
./Libraries/PY32F0xx_LL_Driver/Inc/py32f0xx_ll_rcc.h:584:20: note: each undeclared identifier is reported only once for each function it appears in
  584 |   SET_BIT(RCC->CR, RCC_CR_CSSON);
      |                    ^~~~~~~~~~~~
./Libraries/CMSIS/Device/PY32F0xx/Include/py32f0xx.h:175:42: note: in definition of macro 'SET_BIT'
  175 | #define SET_BIT(REG, BIT)     ((REG) |= (BIT))
      |                                          ^~~
./Libraries/PY32F0xx_LL_Driver/Inc/py32f0xx_ll_rcc.h: In function 'LL_RCC_HSE_EnableBypass':
./Libraries/PY32F0xx_LL_Driver/Inc/py32f0xx_ll_rcc.h:594:20: error: 'RCC_CR_HSEBYP' undeclared (first use in this function); did you mean 'RCC_BDCR_LSEBYP'?
  594 |   SET_BIT(RCC->CR, RCC_CR_HSEBYP);
      |                    ^~~~~~~~~~~~~
./Libraries/CMSIS/Device/PY32F0xx/Include/py32f0xx.h:175:42: note: in definition of macro 'SET_BIT'
  175 | #define SET_BIT(REG, BIT)     ((REG) |= (BIT))
      |                                          ^~~
./Libraries/PY32F0xx_LL_Driver/Inc/py32f0xx_ll_rcc.h: In function 'LL_RCC_HSE_DisableBypass':
./Libraries/PY32F0xx_LL_Driver/Inc/py32f0xx_ll_rcc.h:604:22: error: 'RCC_CR_HSEBYP' undeclared (first use in this function); did you mean 'RCC_BDCR_LSEBYP'?
  604 |   CLEAR_BIT(RCC->CR, RCC_CR_HSEBYP);
      |                      ^~~~~~~~~~~~~
./Libraries/CMSIS/Device/PY32F0xx/Include/py32f0xx.h:177:43: note: in definition of macro 'CLEAR_BIT'
  177 | #define CLEAR_BIT(REG, BIT)   ((REG) &= ~(BIT))
      |                                           ^~~
./Libraries/PY32F0xx_LL_Driver/Inc/py32f0xx_ll_rcc.h: In function 'LL_RCC_HSE_Enable':
./Libraries/PY32F0xx_LL_Driver/Inc/py32f0xx_ll_rcc.h:614:20: error: 'RCC_CR_HSEON' undeclared (first use in this function); did you mean 'RCC_CR_HSION'?
  614 |   SET_BIT(RCC->CR, RCC_CR_HSEON);
      |                    ^~~~~~~~~~~~
./Libraries/CMSIS/Device/PY32F0xx/Include/py32f0xx.h:175:42: note: in definition of macro 'SET_BIT'
  175 | #define SET_BIT(REG, BIT)     ((REG) |= (BIT))
      |                                          ^~~
./Libraries/PY32F0xx_LL_Driver/Inc/py32f0xx_ll_rcc.h: In function 'LL_RCC_HSE_Disable':
./Libraries/PY32F0xx_LL_Driver/Inc/py32f0xx_ll_rcc.h:624:22: error: 'RCC_CR_HSEON' undeclared (first use in this function); did you mean 'RCC_CR_HSION'?
  624 |   CLEAR_BIT(RCC->CR, RCC_CR_HSEON);
      |                      ^~~~~~~~~~~~
./Libraries/CMSIS/Device/PY32F0xx/Include/py32f0xx.h:177:43: note: in definition of macro 'CLEAR_BIT'
  177 | #define CLEAR_BIT(REG, BIT)   ((REG) &= ~(BIT))
      |                                           ^~~
./Libraries/PY32F0xx_LL_Driver/Inc/py32f0xx_ll_rcc.h: In function 'LL_RCC_HSE_IsReady':
./Libraries/PY32F0xx_LL_Driver/Inc/py32f0xx_ll_rcc.h:634:30: error: 'RCC_CR_HSERDY' undeclared (first use in this function); did you mean 'RCC_CR_HSIRDY'?
  634 |   return ((READ_BIT(RCC->CR, RCC_CR_HSERDY) == (RCC_CR_HSERDY)) ? 1UL : 0UL);
      |                              ^~~~~~~~~~~~~
./Libraries/CMSIS/Device/PY32F0xx/Include/py32f0xx.h:179:41: note: in definition of macro 'READ_BIT'
  179 | #define READ_BIT(REG, BIT)    ((REG) & (BIT))
      |                                         ^~~
./Libraries/PY32F0xx_LL_Driver/Inc/py32f0xx_ll_rcc.h: In function 'LL_RCC_HSE_SetFreqRegion':
./Libraries/PY32F0xx_LL_Driver/Inc/py32f0xx_ll_rcc.h:650:26: error: 'RCC_ECSCR_HSE_FREQ' undeclared (first use in this function); did you mean 'RCC_ICSCR_HSI_TRIM'?
  650 |   MODIFY_REG(RCC->ECSCR, RCC_ECSCR_HSE_FREQ, HSEFreq);
      |                          ^~~~~~~~~~~~~~~~~~
./Libraries/CMSIS/Device/PY32F0xx/Include/py32f0xx.h:183:41: note: in definition of macro 'WRITE_REG'
  183 | #define WRITE_REG(REG, VAL)   ((REG) = (VAL))
      |                                         ^~~
./Libraries/PY32F0xx_LL_Driver/Inc/py32f0xx_ll_rcc.h:650:3: note: in expansion of macro 'MODIFY_REG'
  650 |   MODIFY_REG(RCC->ECSCR, RCC_ECSCR_HSE_FREQ, HSEFreq);
      |   ^~~~~~~~~~
./Libraries/PY32F0xx_LL_Driver/Inc/py32f0xx_ll_rcc.h: In function 'LL_RCC_ClearFlag_HSERDY':
./Libraries/PY32F0xx_LL_Driver/Inc/py32f0xx_ll_rcc.h:1500:22: error: 'RCC_CICR_HSERDYC' undeclared (first use in this function); did you mean 'RCC_CICR_LSERDYC'?
 1500 |   SET_BIT(RCC->CICR, RCC_CICR_HSERDYC);
      |                      ^~~~~~~~~~~~~~~~
./Libraries/CMSIS/Device/PY32F0xx/Include/py32f0xx.h:175:42: note: in definition of macro 'SET_BIT'
[... 114 lines cut ...]
./Libraries/PY32F0xx_LL_Driver/Inc/py32f0xx_ll_usart.h: In function 'LL_USART_ConfigMultiProcessMode':
./Libraries/PY32F0xx_LL_Driver/Inc/py32f0xx_ll_usart.h:1232:26: error: 'USART_CR3_HDSEL' undeclared (first use in this function); did you mean 'USART_CR3_CTSE'?
[... cut ...]
./Libraries/PY32F0xx_LL_Driver/Inc/py32f0xx_ll_usart.h:1943:24: error: 'USART_CR1_SBK' undeclared (first use in this function); did you mean 'USART_CR1_WAKE'?
 1943 |   SET_BIT(USARTx->CR1, USART_CR1_SBK);
      |                        ^~~~~~~~~~~~~
./Libraries/CMSIS/Device/PY32F0xx/Include/py32f0xx.h:175:42: note: in definition of macro 'SET_BIT'
  175 | #define SET_BIT(REG, BIT)     ((REG) |= (BIT))
      |                                          ^~~
make: *** [rules.mk:82: Build/User/main.o] Error 1
```

It does compile great with LIB_FLAGS = PY32F030x6 as before.

prosper00, 2023-04-04:
You are correct. I'd had "LIB_FLAGS       = PY32F002x5" instead of "PY32F002Ax5"

However, this may mean there's something broken with the 002x5 target? Looking at the datasheets, I don't see any documentation for the 'non-A' part, or if it even exists (oddly, puya's website no longer makes any mention of the 002 parts at all). And LCSC only has -A parts listed.

In any case, I'm closing this for now. Can re-open is any 002 parts show up.

Thanks!

closed 2023-04-04 by prosper00

IOsetting, 2023-04-05:
This part 002x5 exists in Puya.PY32F0xx_DFP.1.1.3.pack, but I didn't see it be mentioned in Puya's website or anywhere else. I guess this part once exists in their early product roadmap but was renamed to 002Ax5, the only difference between 002x5 and 002Ax5 is the Comparator, 002x5 has no Comparator.

To avoid confusion, I am going to remove this part number from the repository.