IOsetting/py32f0-template #3: Add HAL I2C example

kholia, 2023-03-01:
I am wondering if it would be possible to add a HAL I2C example.

A HAL I2C example would make translating a bunch of STM32 code examples (like the link below) to PY32 world a bit easier.

https://controllerstech.com/i2c-lcd-in-stm32/

Thanks for the help.

IOsetting, 2023-03-02:
I don't have a PCF8574 at hand, so I wrote a demo of [SSD1306](/IOsetting/py32f0-template/tree/main/Examples/HAL/I2C/SSD1306_128x64OLED) for your reference, the process might be similar.

kholia, 2023-03-02:
Thank you @IOsetting. The `SSD1306` example works perfectly on `PY32F030F28P6TU TSSOP20` chip.

I was able to get the 16x2 I2C LCD working on the same chip as well with minimal effort - all thanks to your new HAL I2C example.

I will open a PR to add the `16x2 I2C LCD` example to the repository.

New challenge for me -> Get both these examples working on `PY32F003W16S6TU - SOP16` chip.

This SOP16 IC is much easier to solder than the TSSOP20 one, and has enough pins to be usable in many applications. Hence my motivation to test stuff on it.

I am attaching the datasheet for this SOP16 chip below:

[PY32F003W16S6TU.pdf](https://github.com/IOsetting/py32f0-template/files/10867951/PY32F003W16S6TU.pdf)

closed 2023-03-02 by kholia

IOsetting, 2023-03-31:
@kholia 

I bought some PY32F003W18S, and met the same problem you mentioned in #4

After struggling for a week, I managed to use PF1 and PF0 to drive 1602 LCD

1. The capacitors connected to PF1 and PF0 should be removed. I always add a 100nF capacitor to the reset button when design the PCB, but in this case, this capacitor ruined the SDA signal.
1. Change the option bytes to disable RESET function on PF2
    This step is must, but PY32F003W18S has no BOOT0 pin, this can only be done in the code or manually edit in JLink/DAPLink
1. Change pin PA14 and PF2 to analog mode, this is mentioned in the datasheet.

I have added the demo code of PY32F003W18S to [examples/LL/I2C/PCF8574_1602LCD_PY32F003W_PF0_PF1](/IOsetting/py32f0-template/tree/main/Examples/LL/I2C/PCF8574_1602LCD_PY32F003W_PF0_PF1)

[image]

kholia, 2023-04-01:
Thank you @IOsetting!

Before going a a break, I was able to get the 16x2 LCD display working with `PY32F003W16S6TU - SOP16` almost 2-3 weeks ago. While the code from this repository worked on one display, it wouldn't initialize another display (purchased from another vendor) at all! This issue was pretty bizarre.

What helped in my case was `better / more robust` 16x2 LCD initialization code (to be shared). No hardware changes were required - I don't have any caps connected - not even the 100nF power supply decoupling cap. For actual production usage, I will add the pull-up resistors and decoupling cap(s) as required.

IOsetting, 2023-04-01:
Good to know, have you tried a different I2C address? 
I got 3 1602 LCDs and failed to make two of them work till I found out the address of them are 0x7E. The part marking of these two are PCF8574AT, not PCF8574T, so if I use the address 0x4E, they won't work.

kholia, 2023-04-01:
In my case, the address was same for the two display pieces. One display piece worked with the repository code around 70% of the time, and at other times it failed to initialize. The other display piece (with same address) failed to initialize always. Pretty puzzling for sure. The 100% failing LCD display piece worked just fine with Raspberry Pi Pico.

Once I changed to the new LCD initialization code both display pieces started working with 100% reliability. I haven't traced how this new initialization code is different (yet).

IOsetting, 2023-04-01:
Sounds great, looking forward to your new initialization code :smile:

kholia, 2023-04-02:
I am attaching the `16x2 LCD code` here. I don't recall from where I got it initially (holiday effect!).

[LCD1602_config.h.txt](https://github.com/IOsetting/py32f0-template/files/11132394/LCD1602_config.h.txt)
[LCD1602.h.txt](https://github.com/IOsetting/py32f0-template/files/11132395/LCD1602.h.txt)
[LCD1602.c.txt](https://github.com/IOsetting/py32f0-template/files/11132396/LCD1602.c.txt)

Do give this new library code a try. If it works for you as well, I can open a PR to update the existing 16x2 LCD code we have.

IOsetting, 2023-04-03:
Thank you @kholia, it looks good to me, please create a PR. The initialization steps are similar to [HD44780 datasheet](https://www.sparkfun.com/datasheets/LCD/HD44780.pdf) page 46.