IOsetting/py32f0-template #6: Can support ST-Link V2 to download F200A

Witawat, 2023-03-28:
Can support ST-Link V2 to download F200A
how to setup on windows ?

IOsetting, 2023-03-28:
> Can support ST-Link V2 to download F200A how to setup on windows ?

Yes, but only if you use Keil MDK. This repository is for GNU GCC toolchain(mostly for Linux users) so most of content here might not help.

Keil MDK is a commercial software, you can install their [community edition](https://www2.keil.com/mdk5/selector)

After Keil MDK is installed, you need to install the  DFP pack, which you can download from [Puya PY32F0 product page](https://www.puyasemi.com/cpzx3/info_267_aid_242_kid_235.html), the [download link](https://www.puyasemi.com/uploadfiles/2022/11/PY-MCU%E8%B5%84%E6%96%99-20221117.rar). Then you can use any of JLink/STLink/DAPLink to flash/debug your PY32F002A.

For more details you can read the manuals in that pack
```
PY-MCU资料-20221117$ tree
.
├── Application Note
├── Datasheet&Reference_manual
├── Library and Examples
├── pack
│   ├── IAR
│   └── MDK
│       └── Keil
│           ├── Puya.PY32F0xx_DFP.1.1.0.pack
│           └── PY32F0xx_DFP User Manual Rev.1.0.pdf   <--- this
├── PuyaISP
├── PuyaProgrammer
├── PY_LINK用户手册V1.0
│   ├── PY-LINK User Manual Rev.1.0_EN.pdf       <---- and this
│   └── PY-LINK User Manual Rev.1.0.pdf
└── Start Kit用户指南
```

IOsetting, 2023-03-29:
Thanks for the links.

The JLink OB I am using looks like this one,  
https://www.aliexpress.us/item/3256803434277250.html

The DAPLink I am using is actually an AIR32F103 board([link](https://www.aliexpress.us/item/3256804448776969.html)) with DAPLink firmware ([link](https://github.com/openLuat/luatos-soc-air32f103/tree/master/AIR_Jlink_Keil)), they just released the [source code](https://gitee.com/openLuat/daplink). This is not a genuine STM32, but it works :wink:

prosper00, 2023-03-29:
there might be something fishy with my setup. I was playing around with some of the non-working probes, and while they all show up when I do a 'pyocd list,' none of them (save for one) work. They all generate varying errors with pyocd, and likewise generate unpredictable results with 'openocd -f cmsis-dap.cfg' The closest to a working probe is this one I built according to https://github.com/wagiminator/CH552-picoDAP - it *erases* the py32 just fine, but it times out when trying to write a firmware.

In any case, I'm just glad to have one probe that works. Otherwise, I suspect some sort of USB timing issues.

prosper00, 2023-03-31:
I spent some time documenting my results with a handful of different probes (and probe firmwares), and opened an [issue](https://github.com/pyocd/pyOCD/issues/1523) with pyOCD.

IOsetting, 2023-04-01:
Awesome! Thank you for spending so much time investigating this issue.

kholia, 2023-04-04:
Just dropping a quick note:

I couldn't get my J-Link to talk to PY32 chips - this was my first time using a J-Link.

I also ran into some of the issues as noted by @prosper00.

What worked for me:

Use https://github.com/RadioOperator/STM32F103C8T6_CMSIS-DAP_SWO along with a cheap, easily available STM32 BluePill board. This firmware and hardware combination works with pyOCD very well without any problems.

Also, https://daplink.io/ looked promising but it wouldn't work with the STM32 BluePill board.

I can open a PR to document my working programming setup.

prosper00, 2023-04-04:
I did try a bluepill, with this firmware: https://github.com/devanlai/dap42 - I also flashed it onto an old stlink clone (which, incidentally, make great little usb dev boards cheap). Neither worked with pyocd with any target board. They were able to flash an stm32 target from openocd though. I've been trying to put together a target .cfg file for the py32 and openocd, with (very) limited success so far

kholia, 2023-04-04:
Yep - https://github.com/devanlai/dap42 didn't work me as well.

https://github.com/RadioOperator/STM32F103C8T6_CMSIS-DAP_SWO works like a champ on the BluePill board.

prosper00, 2023-04-04:
oh yeah - the RadioOperator version works for me in pyocd with the PY32 too.  Ironically, I used the dap42 programmer with openocd to flash it to the bluepill...

Tomorrow I'll flash it to an stlink and put it in my bin of dongles. 

Makes me really wonder what's up with pyocd, and want to double down on getting openocd up and running. Especially as openocd seems to be about 40 to 50 times faster than pyocd

kholia, 2023-04-04:
> Yep - https://github.com/devanlai/dap42 didn't work me as well.

I have a very hacky pyOCD patch to make dap42 work with PY32. I just ignore some ACK related errors in pyOCD...

prosper00, 2023-04-24:
> The JLink OB I am using looks like this one, https://www.aliexpress.us/item/3256803434277250.html

Can confirm - the linked JLinkOB clone works for me. I also bought a genuine JLink Edu. It works too.

prosper00, 2023-05-01:
Something I just discovered: programming works better at 3V3 Vcc. At least some of the probes that weren't working for me actually DO work, but only when running the PY32 at 3.3V. And all probes seem to work with more stability and reliability at 3.3V

kholia, 2023-05-02:
Nice detective work :+1: 

This may be because the probes themselves are 3V3 devices?

IOsetting, 2023-05-04:
That's is good to know. I rarely run py32 under 5V, so I didn't notice the difference.

prosper00, 2023-05-15:
I'd be curious to see your openocd fork. I got as far as trying to create a flash routine in src/flash/nor/ ... never did get it to work. Getting it to connect and debug a running program, yes, but I gave up on trying to get flash working.

In retrospect, yeah, using a 3.3V probe with 3.3V I/O was never going to work well, and probably risked damaging the probe too.

NoNamedCat, 2023-06-07:
I think i managed to write to the PY32F002 with a CMSIS_DAP made of a CH552. The cheapest CMSIS_DAP i think?

NoNamedCat, 2023-06-07:
[image]

NoNamedCat, 2023-06-07:
This is the firmware for the CMSIS_DAP based on the CH552:
https://github.com/DeqingSun/ch55xduino/tree/ch55xduino/ch55xduino/ch55x/libraries/Generic_Examples/examples/05.USB/CMSIS_DAP

NoNamedCat, 2023-06-07:
I flashed the blink a led app but it doesn't flash the led

NoNamedCat, 2023-06-09:
Now it's working. My bad... I modify the Make File with the wrong parameters. I can confirm that you can flash the PY32F002 with a CH55x based CMSIS_DAP.

NoNamedCat, 2023-06-09:
If anyone is interested on the Design and gerber files to make a simple board for the PY32F002  and a cheap board based on the CH552 I can put the files here.

deividAlfa, 2023-07-26:
> If anyone is interested on the Design and gerber files to make a simple board for the PY32F002

I made one here:
[image]

https://www.eevblog.com/forum/microcontrollers/$0-11-py32f002a-m0-24mhz-320kb-actually-324kb-more-peripherals/

Witawat, 2023-09-25:
> > If anyone is interested on the Design and gerber files to make a simple board for the PY32F002
> 
> I made one here: [image]
> 
> https://www.eevblog.com/forum/microcontrollers/$0-11-py32f002a-m0-24mhz-320kb-actually-324kb-more-peripherals/

have all file on github ?

deividAlfa, 2023-09-25:
Are you blind?😄 Files are in the link!

Avtopolet, 2023-12-20:
Did I understand correctly, this makefile will not work under Windows?

IOsetting, 2023-12-21:
> Did I understand correctly, this makefile will not work under Windows?

You need something like mingw64 to run make under windows.

Avtopolet, 2023-12-21:
Thanks for answer. Im try it. GNU make output:
PS C:\PUYA\py32f0-template-main> make     
File not found - *.c
File not found - *.c
File not found - *.c
File not found - *.s
process_begin: CreateProcess(C:\xpack\xpack-arm-none-eabi-gcc-13.2.1-1.1\bin\arm-none-eabi-gcc, C:/xpack/xpack-arm-none-eabi-gcc-13.2.1-1.1/bin/arm-none-eabi-gcc -dumpversion, ...) failed.
'expr' is not recognized as an internal or external command,
operable program or batch file.
'tr' is not recognized as an internal or external command,
operable program or batch file.
'' was unexpected at this time.
make: *** [fullcheck] Error 255

IOsetting, 2023-12-21:
Download Arm GCC toolchain from https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads and extract it to /opt,

```bash
# check version
ls /opt/arm-gnu-toolchain-13.2.Rel1-mingw-w64-i686-arm-none-eabi/bin/arm-none-eabi-gcc
/opt/arm-gnu-toolchain-13.2.Rel1-mingw-w64-i686-arm-none-eabi/bin/arm-none-eabi-gcc --version

arm-none-eabi-gcc.exe (Arm GNU Toolchain 13.2.rel1 (Build arm-13.7)) 13.2.1 20231009
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

# unzip the code
unzip py32f0-template-main.zip
cd py32f0-template-main/

# install nano editor
pacman -S nano

# edit Makefile
nano Makefile
```
Update the toolchain path in Makefile
```
##### Toolchains #######
ARM_TOOCHAIN    ?= /opt/arm-gnu-toolchain-13.2.Rel1-mingw-w64-i686-arm-none-eabi/bin
```
Then make clean and make, the console output
```bash
iosetting@a02-NC01 MSYS ~/py32f0-template-main
$ make clean
rm -rf ./Build/*

iosetting@a02-NC01 MSYS ~/py32f0-template-main
$ make
  CC    User/main.c
  CC    User/py32f0xx_hal_msp.c
  CC    User/py32f0xx_it.c
  CC    Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal.c
  CC    Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_adc.c
  CC    Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_adc_ex.c
  CC    Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_comp.c
  CC    Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_cortex.c
  CC    Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_crc.c
  CC    Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_dma.c
  CC    Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_exti.c
  CC    Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_flash.c
  CC    Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_gpio.c
  CC    Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_i2c.c
  CC    Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_irda.c
  CC    Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_iwdg.c
  CC    Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_led.c
  CC    Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_lptim.c
  CC    Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_pwr.c
  CC    Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_rcc.c
  CC    Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_rcc_ex.c
  CC    Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_rtc.c
  CC    Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_rtc_ex.c
  CC    Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_spi.c
  CC    Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_tim.c
  CC    Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_tim_ex.c
  CC    Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_uart.c
  CC    Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_usart.c
  CC    Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_wwdg.c
  CC    Libraries/PY32F0xx_HAL_BSP/Src/py32f0xx_bsp_clock.c
  CC    Libraries/PY32F0xx_HAL_BSP/Src/py32f0xx_bsp_led.c
  CC    Libraries/PY32F0xx_HAL_BSP/Src/py32f0xx_bsp_printf.c
  CC    Libraries/CMSIS/Device/PY32F0xx/Source/system_py32f0xx.c
  AS    Libraries/CMSIS/Device/PY32F0xx/Source/gcc/startup_py32f030.s
  LD    Libraries/LDScripts/py32f030x8.ld -> Build/app.elf
Memory region         Used Size  Region Size  %age Used
             RAM:        2280 B         8 KB     27.83%
           FLASH:        7808 B        64 KB     11.91%
  OBJCP BIN     Build/app.bin
  OBJCP HEX     Build/app.hex

iosetting@a02-NC01 MSYS ~/py32f0-template-main
```

Avtopolet, 2023-12-22:
Thanks for Help!!!  My setup - win10, STlink v2 (clone) converted to JLink, processor from disposable vape.
xxx@V55T-1 MINGW64 /c/PUYA/py32f0-template-main
$ make flash
pyocd erase -t py32f002ax5 --chip --config ./Misc/pyocd.yaml
0000966 I Erasing chip... [eraser]
0001274 I Chip erase complete [eraser]
pyocd load ./Build/app.hex -t py32f002ax5 --config ./Misc/pyocd.yaml
0000939 I Loading C:\PUYA\py32f0-template-main\Build\app.hex [load_cmd]
[==================================================] 100%
0004147 I Erased 8192 bytes (2 sectors), programmed 5632 bytes (44 pages), skipp
ed 0 bytes (0 pages) at 1.72 kB/s [loader]

IOsetting, 2023-12-22:
I just googled to check what is "disposable vape" and why these is a processor in it, very interesting, I didn't know this kind of e-cig before.

kholia, 2024-11-15:
https://github.com/raspberrypi/debugprobe works really well with pyOCD and is also well maintained - highly recommended!