IOsetting/py32f0-template #8: Does PY32Fxx series support True Open Drain?

llinjupt, 2023-04-16:
Hi IOsetting.

I wonder if PY32Fxx series support True Open Drain when with 3V3 supply like STM8S does. I have read the datasheet and nothing about 5VFT. It seems that you are so familiar with this series. Would you like to confirm if it's supported. Thanks in advance.

IOsetting, 2023-04-18:
Hi llinjupt,

I got no technical support from Puya so I cannot give a definite answer. According to the voltage limits described in *表 5-1 电压特性* in the datasheet, if the numbers are true, these pins are not true open drain, and they are not 5V tolerant when the power supply is 3.3V.

llinjupt, 2023-04-25:
@IOsetting  Thanks. Arrording to the table, the maximum input voltage is VCC+0.3, so no pin  is 5V tolerant 
when with 3V3 supply.