IOsetting/py32f0-template #10: PY32F002AF15P6TU Breakout board

NoNamedCat, 2023-05-25:
This is a simple dev board that I design... It has jumpers to select the VCC voltage, it hasonboard 3.3v, boot and reset buttons, and the elements for a future softwrare usb??

[image]
[image]

NoNamedCat, 2023-05-25:
This board was done in Kicad. You can use it without giving me credit. It's a gif :)
[PY32F002.zip](https://github.com/IOsetting/py32f0-template/files/11562954/PY32F002.zip)

NoNamedCat, 2023-05-25:
This is the bom for LCSC
[bom.csv](https://github.com/IOsetting/py32f0-template/files/11563430/bom.csv)

IOsetting, 2023-05-25:
Nice design!

I happen to have ordered several PY32F0xx boards on JLCPCB, too. I am not good at KiCad so I just used their online EDA tool [lceda.cn](https://lceda.cn/). These boards are open sourced on https://oshwhub.com/

* PY32F002A https://oshwhub.com/iosetting/py32f002a-tssop20-evb
* PY32F003F1x https://oshwhub.com/iosetting/py32f003f1x
* PY32F003F2x(no osc) https://oshwhub.com/iosetting/py32f003f2x
* PY32F003F2x https://oshwhub.com/iosetting/py32f003f2x-osc
* PY32F003W SOP16 https://oshwhub.com/iosetting/py32f003w
* PY32F030F1x https://oshwhub.com/iosetting/py32f030f1x

NoNamedCat, 2023-05-25:
If you want I can pass to you the Gerber files plus I can do a custom silkscreen or modifications for free for this repo. I just want to tank u and make something for the project

IOsetting, 2023-05-25:
Thank you for your kindness! I am wondering if you can create a repository for this PCB design, then I will add a link in README. 
In this way you will be able to update the design anytime you like.

NoNamedCat, 2023-05-25:
You can add the design to this repo. I really don't want any mentions, you don't have to give me credits or anything... The design is completely yours

NoNamedCat, 2023-05-31:
Hello again. Do you want to redesign the boards that you posted before?
I can make some designs for you if you want.

NoNamedCat, 2023-06-07:
[image]

NoNamedCat, 2023-06-07:
Sorry, in the moment i only have a potatoe to take photos