IOsetting/py32f0-template #12: Bit-Banged usb?

NoNamedCat, 2023-06-06:
Do you think that bit-banging USB 1.0 is possible with this MCU?
I did a quick search of repositories on github and I found this:
[https://github.com/ads830e/stm32f030-vusb](url)
Theoretically it would be possible to port this code to these MCUs, the library is made for microcontrollers of the STM brand in assembly and runs at 48mhz. The core of the chip for which this library was made is also Cortex M0

NoNamedCat, 2023-06-06:
BTW the boards that i designed just arrived today. Tomorrow I will solder some of them and I will test the design.

IOsetting, 2023-06-07:
That's impressive! I guess it is possible, considering that the emulation appears to be based on GPIO and EXTI.

cnlohr, 2025-04-21:
For ARM based systems, grainuum is much more for you even though it is much less complete than rv003usb.  

One warning, you will need to operate at 48MHz

NoNamedCat, 2025-04-25:
But graniuum is not very complete, doesn't have many examples as the rv003usb. I think that if is possible to port the core of the rv003usv to the py32f002 the benefits of the port can make a big boost to the community.

cnlohr, 2025-04-26:
Does the py32f002 operate reliably at 48MHz?   If so you have a shot.  If not, it would be virtually impossible to do in a generic way that does not have a lot of compromises that would be really painful.