IOsetting/py32f0-template #13: 新人求助 关于ll库

stimjoe, 2023-06-07:
我在makefile里启用了LL库,但是在程序中使用BSP的还是HAL库的,使用LL库的内容也提示错误
[image]
[image]

`# Use LL library instead of HAL, y:yes, n:no
USE_LL_LIB ?= y
# Enable printf float %f support, y:yes, n:no
ENABLE_PRINTF_FLOAT	?= n
# Build with FreeRTOS, y:yes, n:no
USE_FREERTOS	?= n
# Build with CMSIS DSP functions, y:yes, n:no
USE_DSP			?= n
# Build with Waveshare e-paper lib, y:yes, n:no
USE_EPAPER		?= n
# Programmer, jlink or pyocd
FLASH_PROGRM	?= pyocd`

IOsetting, 2023-06-07:
The syntax errors are solely a matter of display and do not affect compilation. If you wish to resolve the syntax errors, please investigate the VSCode settings and the .vscode/c_cpp_properties.json file to identify possible causes.

Enabling the LL library in the Makefile is not sufficient. 

To resolve the issue, kindly replace the default files (which are intended for the HAL library) under the User folder with the examples provided in the Examples/LL directory.

stimjoe, 2023-06-07:
i added a line in .vscode/c_cpp_properties.json defines as USE_FULL_LL_DRIVER ,and rename \Libraries\BSP_LL to \Libraries\BSP 
the issue is solved.Thanks for your help.

closed 2023-06-07 by stimjoe