IOsetting/py32f0-template #15: What is the closest STM32 part¿?

NoNamedCat, 2023-06-28:
Hello again.
Im wondering what is the closest MCU on the STM32 family to this microcontroller?
I want try to port the arduino core for the PY32F0 and i don't want to start from scratch.

prosper00, 2023-06-28:
the official stmicro arduino cores are pretty terrible. They're built on top of the HAL - which might make them easy to port, but not exactly lightweight. Arduino's on HAL, sitting on top of ST's HAL - not a great fit for smaller arm micros. It's hard getting anything useful to fit in flash with so much blubber. Even just the basic 'blink' example built on Arduino is huge:

for the Nano, blink is 924 bytes with 9 bytes of sram.
for the stm32g030, it's **13304 bytes of flash, and 1236** of SRAM!! for *BLINK*!! They sent a man to the moon with less resources.

It's ridiculous. Especially when the smaller arm cores don't exactly have flash and sram to spare.

NoNamedCat, 2023-06-28:
Some time ago, I found and then forgot where I found an Arduino core that is not based on HAL but on LL. Do you think that would be better? I suppose the STM32G030 is the MCU most similar to the PY32F030, PY32F002...?

NoNamedCat, 2023-06-28:
https://github.com/geosmall/Arduino_STM32_LL?

prosper00, 2023-06-28:
It seems relatively similar by specs, but I haven't compared them all that closely. The peripheral set has a slight resemblance, but it's definitley not a 1:1 clone

IOsetting, 2023-06-29:
> Hello again. Im wondering what is the closest MCU on the STM32 family to this microcontroller? I want try to port the arduino core for the PY32F0 and i don't want to start from scratch.

Someone is already working on this: https://github.com/Air-duino/Arduino-AirMCU

AIR001xx is a variant(much likely the same core) of PY32F002A, so in theory the Arduino implementation also works on PY32F003 and PY32F030.

More details about AIR001xx can be found at below links. This chip hasn't been released officially, some documents may be incomplete
https://gitee.com/openLuat/luatos-broads/tree/master/broads/Air001-Board/
https://wiki.luatos.com/chips/air001/index.html

IOsetting, 2023-06-29:
Thank you for your kindness but I already have a bunch of pcb boards of PY32 chips. Recently my interests have been shifting from PY32. 
I believe that you are professional in pcb design and you have good ideas that you want to share, please create a repository for your design, then you can share and get feedback, that is how GitHub works.

kholia, 2024-11-20:
@NoNamedCat It seems PY32F030 is a mashup of STM32F030 and STM32G0 ("IOPORT" stuff).

At this this is what I found during my https://github.com/kholia/zephyr/tree/Support-for-Puya-PY32 work.

NoNamedCat, 2024-11-20:
There is a port of Arduino for the PY32 microcontrollers already.
https://github.com/py32duino/Arduino-PY32

NoNamedCat, 2024-11-20:
Very interesting the port of ZephirOS to the pu32f003. Tks a lot!

kholia, 2026-11-09:
https://github.com/zephyrproject-rtos/zephyr/pull/124981 is up now - For folks looking to use Zephyr on PY32 chips.

Thanks to @IOsetting for making this possible!