IOsetting/py32f0-template #19: Attaching py32F002 english docs

deividAlfa, 2023-07-09:
I'm attaching the official english documentation, taken from somewhere at the Puya website which I wasn't able to find again.

[PY32F002A datasheet Rev.0.1_EN.pdf](https://github.com/IOsetting/py32f0-template/files/11996042/PY32F002A.datasheet.Rev.0.1_EN.pdf)

[PY32F002A Reference manual v1.0_EN.pdf](https://github.com/IOsetting/py32f0-template/files/11996043/PY32F002A.Reference.manual.v1.0_EN.pdf)

prosper00, 2023-07-09:
d 'ya have english RMs for the other py32's in your collection?

deividAlfa, 2023-07-10:
All docs I got were from Puya website.
Please check there for the what I called "Puya support pack" 1 and 2:

https://www.eevblog.com/forum/microcontrollers/$0-11-py32f002a-m0-24mhz-320kb-actually-324kb-more-peripherals/

Those big files contain all the available docs.
Puya ISP is also very interesting, works great.

prosper00, 2023-07-10:
OH, wait, there *is* english in there. for some reason, i was thinking it was english datasheet only