IOsetting/py32f0-template #22: Possible hardware bug: CCP hangs the system if RDP1 is set ?

deividAlfa, 2023-08-18:
I simplify the project as most as I could.
Sorry, it's made for patched CubeIDE, I'm not good with makefiles, but it uses the same files as this template, so it should be easy to setup.
Timer1 CC ISR captures falling edges, sets  "new_capture" variable and disables itself.
Systick ISR check for this variable, makes a led pulse as indication, also toggles another led to indicate the system is running, and restarts the capture.
Without RDP, I can beat the CCP input in any way, it always works under any optimization level.
When I enable RDP, it will randomly crash.
I attached the debugger, with RDP enabled I can only access the CPU registers, but I can see the program counter.
In the instant the problem happens, the debugger halts and shows this:
```
...Target halted (Vector catch, PC = 0xFFFFFFFE)
Reading common registers: Read register 'r0' (4 bytes) from hardware: 0x0000000A
Read register 'r1' (4 bytes) from hardware: 0x05000000
Read register 'r2' (4 bytes) from hardware: 0x01000000
Read register 'r3' (4 bytes) from hardware: 0x01000000
Read register 'r4' (4 bytes) from hardware: 0xE70B0020
Read register 'r5' (4 bytes) from hardware: 0x01000000
Read register 'r6' (4 bytes) from hardware: 0xDC0BFF1F
Read register 'r7' (4 bytes) from hardware: 0xD80B0020
Read register 'r8' (4 bytes) from hardware: 0x24B763A1
Read register 'r9' (4 bytes) from hardware: 0x64020020
Read register 'r10' (4 bytes) from hardware: 0x1BD00CED
Read register 'r11' (4 bytes) from hardware: 0x273F3419
Read register 'r12' (4 bytes) from hardware: 0x00E0FFFF
Read register 'sp' (4 bytes) from hardware: 0xB80B0020
Read register 'lr' (4 bytes) from hardware: 0xF9FFFFFF
Read register 'pc' (4 bytes) from hardware: 0xFEFFFFFF
Read register 'xpsr' (4 bytes) from hardware: 0x03000061
WARNING: Failed to read memory @ address 0xFFFFFFFE
Read 4 bytes @ address 0x20000BD4 (Data = 0x61000000)
Reading 64 bytes @ address 0x20000BC0
WARNING: Failed to read memory @ address 0x0800080C
WARNING: Failed to read memory @ address 0x08001EFA
WARNING: Failed to read memory @ address 0x08001EFA
Read 4 bytes @ address 0xE000ED04 (Data = 0x00420003)
Read 4 bytes @ address 0xE000ED0C (Data = 0xFA050000)
Read 4 bytes @ address 0xE000ED04 (Data = 0x00420003)
Read 4 bytes @ address 0xE000ED1C (Data = 0x00000000)
Read 4 bytes @ address 0xE000ED20 (Data = 0x00000000)
Read 4 bytes @ address 0xE000ED24 (Data = 0x00000000)
Read 4 bytes @ address 0xE000ED08 (Data = 0x08000000)
Read 4 bytes @ address 0xE000E010 (Data = 0x00000004)
Read 4 bytes @ address 0xE000E014 (Data = 0x000F398E)
Read 4 bytes @ address 0xE000E018 (Data = 0x009F0E6E)
Read 4 bytes @ address 0xE000E01C (Data = 0x80000000)
Read 4 bytes @ address 0xE000E01C (Data = 0x80000000)
```
This is not a HardFault, as I have a led toggling routine there that would flash a led, it works if I try to access unimplemented memory.

Here's the project:
[PY32F002A_test.zip](https://github.com/IOsetting/py32f0-template/files/12378714/PY32F002A_test.zip)

deividAlfa, 2023-08-18:
This the same code, compiled without setting RDP.
Works fine flashing with PuyaISP.
But if RDP is enabled in the PuyaISP setings, it will also fail.

[py32f002a_ccp.zip](https://github.com/IOsetting/py32f0-template/files/12379173/py32f002a_ccp.zip)

MCU:PY32F002AF15P6TU

System LED:  PB1, active low
CCP LED:     PB0, active low
CCP input:   PA1, falling edge, internal pullup.
Oscillator:  Internal 24MHZ HSI

Nothing else is required. Just poke the CCP input and see the CCP led doing stuff.
At some point it will stall.

Leds are active-low, connected to VDD, the mcu pin pull it low when active.

I've sent an email to PuyaSemi because this doesn't make any sense!

deividAlfa, 2023-08-18:
[image]
[image]
[image]

IOsetting, 2023-08-23:
@deividAlfa 
Apologies, I just came back from a trip.  
Thank you for the PR.
I once noticed the special handling in [system_py32f0xx.c](https://github.com/IOsetting/py32f0-template/blob/main/Libraries/CMSIS/Device/PY32F0xx/Source/system_py32f0xx.c) 
```c
#ifndef FORBID_VECT_TAB_MIGRATION
#ifndef VECT_TAB_SRAM
#if (defined (__CC_ARM)) || (defined (__ARMCC_VERSION) && (__ARMCC_VERSION >= 6010050))
extern int32_t $Super$$main(void);
uint32_t VECT_SRAM_TAB[48]__attribute__((section(".ARM.__at_0x20000000")));

/* re-define main function */
int $Sub$$main(void)
{
  uint8_t i;
  uint32_t *pFmcVect = (uint32_t *)(FLASH_BASE | VECT_TAB_OFFSET);
  for (i = 0; i < 48; i++)
  {
    VECT_SRAM_TAB[i] = pFmcVect[i];
  }

  SCB->VTOR = SRAM_BASE;

  $Super$$main();
  return 0;
}
#elif defined(__ICCARM__)
extern int32_t main(void);
/* __low_level_init will auto called by IAR cstartup */
extern void __iar_data_init3(void);
uint32_t VECT_SRAM_TAB[48] @SRAM_BASE;
int __low_level_init(void)
{
  uint8_t i;
  uint32_t *pFmcVect = (uint32_t *)(FLASH_BASE | VECT_TAB_OFFSET);
  /* call IAR table copy function. */
  __iar_data_init3();

  for (i = 0; i < 48; i++)
  {
    VECT_SRAM_TAB[i] = pFmcVect[i];
  }

  SCB->VTOR = SRAM_BASE;

  main();
  return 0;
}
#endif
#endif
#endif
```
But at that time I didn't understand why they do so. In this way no matter `VECT_TAB_SRAM` is defined or not, the vector is always in SRAM.
I guess the bug you encountered might explain this. In GCC compilation this part is ignored so it doesn't work properly when RDP is enabled.

deividAlfa, 2023-08-27:
Yeah, I also found this code to be strange at best!
I kept asking myself why these compilers needed the NVIC table placed in SRAM... I thought it was a compiler thing.

But now all makes sense. Why didn't they fix it for GCC in first place?
I have no idea, I guess it's the vendor support you get at $0.11 per chip.

I'll take the issue as solved, closing now!

closed 2023-08-27 by deividAlfa