IOsetting/py32f0-template #26: Arduino Support

NoNamedCat, 2023-08-29:
Hello, thank you for this project once again.
I want to add Arduino support for the PY32F002.
As IOsetting mentioned earlier, there are some projects that do this for another chip called AIR001.
I've collected some test sketches and can confirm that if I upload the hexadecimal to the PY32F002, it works perfectly.
So, I'm thinking of cloning the repository and adapting the source code for this MCU.
I have some questions about how to do this.
Is it correct to take the source code from the repository and adapt it for this MCU?
Would any of you be interested in doing this?
Is it worth doing this?
Some of you have mentioned that the resources of this MCU might not be sufficient for the Arduino environment.
Based on the results and libraries ported by the other project, do you believe this to be the case?
Thank you and warm regards!

NoNamedCat, 2023-08-29:
This is the site for the project:

https://wiki.luatos.com/chips/air001/index.html

IOsetting, 2023-08-30:
I seldom use Arduino so I don't have an idea, maybe you can fork it and make further changes to make it support PY32 types?
BTW I don't like their ISP tool because it need .NET runtime, I don't use .NET in Ubuntu. If I got time I would write one in C.

NoNamedCat, 2023-09-01:
I'm in a rabbit hole....

'''c++
#include <stdio.h>
#include <stdlib.h>
#include <hidapi/hidapi.h>
#include <unistd.h>
#include <string.h>

#define REPORT_SIZE 64
#define RETRY_DELAY_SECONDS 5

// Function to send data to the CMSIS-DAP adapter
void send_data(hid_device *device, const unsigned char *data, size_t size) {
    int bytes_sent = hid_write(device, data, size);
    if (bytes_sent < 0) {
        printf("Error sending data\n");
        exit(-1);
    }
}

void print_progress_bar(float progress) {
    int bar_width = 50;
    int pos = bar_width * progress;
    printf("[");
    for (int i = 0; i < bar_width; ++i) {
        if (i < pos) printf("=");
        else printf(" ");
    }
    printf("] %.2f%%\r", progress * 100);
    fflush(stdout);
}

int main(int argc, char *argv[]) {
    if (argc != 2) {
        printf("Usage: %s <file_path>\n", argv[0]);
        return -1;
    }

    const char *file_path = argv[1];
    const char *id_list_file_path = "./ids.txt"; // Corrected path

    hid_device *device;
    unsigned char buf[REPORT_SIZE];

    // Initialize HIDAPI
    if (hid_init() != 0) {
        printf("Failed to initialize HIDAPI\n");
        return -1;
    }
    printf("HIDAPI initialized\n");

    // Open the ID list file for reading
    FILE *id_list_file = fopen(id_list_file_path, "r");
    if (!id_list_file) {
        printf("Failed to open ID list file %s\n", id_list_file_path);
        hid_exit();
        return -1;
    }
    printf("ID list file opened: %s\n", id_list_file_path);

    // Read USB vendor and product IDs from the list file
    unsigned int vendor_id, product_id;
    while (fscanf(id_list_file, " %x %x", &vendor_id, &product_id) == 2) {
        bool device_found = false;
        int retry_count = 0;

        while (!device_found && retry_count < RETRY_DELAY_SECONDS) {
            // Open the CMSIS-DAP adapter with the specified IDs
            device = hid_open(vendor_id, product_id, NULL);
[... 103 lines cut ...]
        if (!verification_failed) {
            printf("Verification succeeded: Flashed data matches original file\n");
        } else {
            printf("Verification failed: Flashed data does not match original file\n");
        }

        // Reset the microcontroller
        buf[0] = 0x00; // Replace with the reset command
        send_data(device, buf, 1);
        printf("Microcontroller reset\n");

        // Close the device
        hid_close(device);
        printf("Device closed\n");
    }

    // Close the ID list file and cleanup HIDAPI (same as before)
    fclose(id_list_file);
    printf("ID list file closed\n");
    hid_exit();
    printf("HIDAPI cleaned up\n");

    return 0;
}
'''

NoNamedCat, 2023-09-01:
I'm in a rabbit hole right now...

I'm trying to make a program that interface with the USB-DAP adapter...

```c++

#include <stdio.h>
#include <stdlib.h>
#include <hidapi/hidapi.h>
#include <unistd.h>
#include <string.h>

#define REPORT_SIZE 64
#define RETRY_DELAY_SECONDS 5

// Function to send data to the CMSIS-DAP adapter
void send_data(hid_device *device, const unsigned char *data, size_t size) {
    int bytes_sent = hid_write(device, data, size);
    if (bytes_sent < 0) {
        printf("Error sending data\n");
        exit(-1);
    }
}

void print_progress_bar(float progress) {
    int bar_width = 50;
    int pos = bar_width * progress;
    printf("[");
    for (int i = 0; i < bar_width; ++i) {
        if (i < pos) printf("=");
        else printf(" ");
    }
    printf("] %.2f%%\r", progress * 100);
    fflush(stdout);
}

int main(int argc, char *argv[]) {
    if (argc != 2) {
        printf("Usage: %s <file_path>\n", argv[0]);
        return -1;
    }

    const char *file_path = argv[1];
    const char *id_list_file_path = "./ids.txt"; // Corrected path

    hid_device *device;
    unsigned char buf[REPORT_SIZE];

    // Initialize HIDAPI
    if (hid_init() != 0) {
        printf("Failed to initialize HIDAPI\n");
        return -1;
    }
    printf("HIDAPI initialized\n");

    // Open the ID list file for reading
    FILE *id_list_file = fopen(id_list_file_path, "r");
    if (!id_list_file) {
        printf("Failed to open ID list file %s\n", id_list_file_path);
        hid_exit();
        return -1;
    }
    printf("ID list file opened: %s\n", id_list_file_path);

    // Read USB vendor and product IDs from the list file
    unsigned int vendor_id, product_id;
    while (fscanf(id_list_file, " %x %x", &vendor_id, &product_id) == 2) {
        bool device_found = false;
        int retry_count = 0;

[... 106 lines cut ...]
        if (!verification_failed) {
            printf("Verification succeeded: Flashed data matches original file\n");
        } else {
            printf("Verification failed: Flashed data does not match original file\n");
        }

        // Reset the microcontroller
        buf[0] = 0x00; // Replace with the reset command
        send_data(device, buf, 1);
        printf("Microcontroller reset\n");

        // Close the device
        hid_close(device);
        printf("Device closed\n");
    }

    // Close the ID list file and cleanup HIDAPI (same as before)
    fclose(id_list_file);
    printf("ID list file closed\n");
    hid_exit();
    printf("HIDAPI cleaned up\n");

    return 0;
}
```

NoNamedCat, 2023-09-01:
I don't know much about USB HID protocol and the DAP protocol...

The binary file needs a file called "ids.txt" on the same folder of the compiled file.

You use the program like this:
./flasher ./fileToFlash.bin

NoNamedCat, 2023-09-01:
This is the makefile:

CC = g++
CFLAGS = -std=c++11 -Wall
LIBS = -lhidapi-libusb

TARGET = flasher

all: $(TARGET)

$(TARGET): flasher.cpp
	$(CC) $(CFLAGS) -o $(TARGET) flasher.cpp $(LIBS)

clean:
	rm -f $(TARGET)

NoNamedCat, 2023-09-01:
The program is not able to flash the microcontroller, test the connection with it, or test if the microcontroller have the correct flash information.

NoNamedCat, 2023-09-01:
Im using the raw values of the file: 
https://github.com/IOsetting/py32f0-template/blob/main/Misc/SVD/py32f002axx.svd
Am I missing something?

IOsetting, 2023-09-02:
@NoNamedCat I don't know what is going on, I didn't see any new replies in this thread, but I found a lot new replies in my mailbox. Regarding the USB-DAP adapter problem, I don't know much about USB HID either, maybe you can find help in [DAPLink discussions](https://github.com/ARMmbed/DAPLink/discussions)

NoNamedCat, 2023-09-03:
Tks IOsetting. I'm trying to make a standalone program to interface with a CMIS-DAP adapter to integrate this into the Arduino IDE. I'm reading about HID, DAP protocol, and I'm trying to communicate using directly with the protocol. For now I'm using WireShark on Linux to try to reverse engineer the steps needed to flash the program on the py32f002.
Tks a lot and sorry for the spam on your email.

NoNamedCat, 2023-09-04:
https://www.keil.com/pack/doc/CMSIS_Dev/General/html/index.html

Here are some resources. I posted here to future reference.

NoNamedCat, 2023-11-13:
https://github.com/py32duino/Arduino-PY32

Arduino for the PY32F series is real now :)

IOsetting, 2023-11-13:
That's great news, the Arduino support will definitely attract more users.

NoNamedCat, 2023-11-13:
Totally agree. That makes my dreams come true. I really want the cheapest possible arduino compatible board.
Here... I give you another design. It has a CH340X for automatic reset and boot mode entering and flasing.

https://github.com/wagiminator/MCU-Flash-Tools/issues/2

The same with the older design. You can use the design as you wish. You can use it in this web page. You can sell it. You can do what you want, You don't need to credit me for that.

The board should be compatible with the puysisp tool that wagiminator created.

NoNamedCat, 2023-11-13:
Should i close this issue?

IOsetting, 2023-11-13:
I don't mind if it is open, but if you like, yes, of course you can close it.

closed 2023-11-14 by NoNamedCat