IOsetting/py32f0-template #30: reprogramming PY32F002A

ajaybhargav, 2023-09-09:
Hi, I have successfully build and flashed PY32F002AL15S, using puyocd and ST-LINK V2. However I am unable to get the IC detected again on SWD. I believe since pins are used inside the firmware (which is already flashed), How can I enable SWD again on this chip? as per reference manual, POR should enable those pins, but pyocd does not detect. I do not have any idea how to reprogram this chip.

IOsetting, 2023-09-09:
Please refer to this discussion [https://www.eevblog.com/forum/microcontrollers/hc32l110-complete-erase/](https://www.eevblog.com/forum/microcontrollers/hc32l110-complete-erase/)
You need a Win10 machine to run the ISP Tool.

IOsetting, 2023-09-09:
Sorry I made a mistake. It is different part. You need to use Puya's ISP Tool [download link](https://www.puyasemi.com/uploadfiles/2022/11/PY-MCU%E8%B5%84%E6%96%99-20221117.rar)
Please refer to this discussion [https://www.eevblog.com/forum/microcontrollers/$0-11-py32f002a-m0-24mhz-320kb-actually-324kb-more-peripherals/](https://www.eevblog.com/forum/microcontrollers/$0-11-py32f002a-m0-24mhz-320kb-actually-324kb-more-peripherals/)

ajaybhargav, 2023-09-09:
There is no boot0 pin on this chip, its an 8 pin IC. I made NRST as GPIO. there is no way to invoke bootloader or enter into flash mode. I think I bricked it 😝 I will try ISP though but as per Puya PY32F002AL15S does not have ISP feature.

IOsetting, 2023-09-10:
It seems only TSSOP-20 package has BOOT0 pin. I did some searching but didn't find any solution. I usually add one second delay before the pin remapping so the chip can still be programmed via SWD after reset.

ajaybhargav, 2023-09-10:
I did not realize I would run into such problem, I was just testing this chip. I am going to keep NRST as is without mapping it as full GPIO, I believe it can be used as GPIO (PF2) (as per manual) after 20 clock cycles have passed.

> I usually add one second delay before the pin remapping so the chip can still be programmed via SWD after reset.

Option byte is a different story, its like transforming your pin to a different function permanently. so software delay has no effect. PF2 is changed to PA2 permanently.

NoNamedCat, 2023-09-10:
Did you read this forum?
https://www.elektroda.com/rtvforum/topic3946116-30.html

"I accidently messed around with A13. Is there any way to 'unbrick'/erase a device before the firmware is loaded? Couldn't find anything in the documentation.

A manual reset quickly following by a erase does not help..
Thanks!

Thank you for your quick reply, I fixed it.

I should have mentioned this, it was the QFN (PY32F002AW15U6TR) giving me problems it has no BOOT0 pin. BOOT0 did help to erase the TSSOP20. Solution for the QFN is to do hardware RESET when powered on.

All very obvious. I made stupid mistake though, I had two projects with the same name but both with different MCU's. I had the wrong processor selected, ouch!"

NoNamedCat, 2023-09-10:
If you have a button connected to the NRST pin, try to hold it at the same time that you power the board. That is what I understand from that forum.

ajaybhargav, 2023-09-10:
I already tried everything, I shorted NRST with GND, it doesnt work. Since I have written the option bytes to change NRST(PF2) to PA2.

PA13 is just SWD pin, if your NRST is intact you can keep CPU in reset by shorting it to GND. if you have Boot0 pin you can put CPU in bootloader. but this 8-bit part has only NRST and that too is now a GPIO, so unless I had some software logic implemented to revert back A2 to F2-NRST, which I did not thought of, since as per reference manual, I thought NRST will only switch once CPU is passes the POR state.

[image]

So yes, there is a possibility of bricking (only 8-pin) Puya.

However, as per Puya FAE; the only way to bring it back is using an offline programmer, looks like the NRST before 20uS filter can only be triggered using and an offline programmer.

deividAlfa, 2023-09-12:
Well you know now: Be grateful, you only lost 10 cents LOL.

fred-corp, 2024-10-31:
How is it possible to force pin 6 to stay the nRST pin on SOP-8 Packages ?

Even if I do,'t use the pin as a GPIO, I have to revive the chip by erasing it with the J-Link Flash app and a wire to reset the chip, otherwise I get a ```Failed to power up DAP```error when trying to re-flash

fred-corp, 2024-10-31:
Found a way to tell the J-Link to force-reset the chip : 

add ```r0``` at the top of ```/Misc/jlink-command``` file

```txt
r0
erase
loadfile Build/app.hex 0 noreset
reset
exit
```

closed 2025-03-06 by ajaybhargav