IOsetting/py32f0-template #32: PY32F002B-PY32F040-PY32F072-PY32F403 PDF Files

NoNamedCat, 2023-10-24:
Here are some PDF's that you can add to the repo. Also there is a new puya M4 MCU (PY32F403). O.o

[Py32F002B.pdf](https://github.com/IOsetting/py32f0-template/files/13112830/Py32F002B.pdf)
[PY32F040.pdf](https://github.com/IOsetting/py32f0-template/files/13112831/PY32F040.pdf)
[Py32F072.pdf](https://github.com/IOsetting/py32f0-template/files/13112832/Py32F072.pdf)
[PY32F403.pdf](https://github.com/IOsetting/py32f0-template/files/13112833/PY32F403.pdf)

IOsetting, 2023-10-24:
Thank you! I have been using [WIKI](https://github.com/IOsetting/py32f0-template/wiki) to organize these documents. 
The Docs folder will be removed in the future to reduce the repository size.

NoNamedCat, 2023-10-25:
Can I close this issue?

IOsetting, 2023-10-25:
Yes, of course.

closed 2023-10-25 by NoNamedCat