IOsetting/py32f0-template #33: Deep sleep current consumption

garudaonekh, 2023-11-25:
Hi, 
I plan to use it for remote sensor with 3 aaa batteries.
I cannot find any info regarding its current consumption in deep sleep.

thanks

IOsetting, 2023-11-27:
In PY32F030 datasheet, there is a table of sleep mode current, these numbers should work for all 002A/003/030.
[image].

garudaonekh, 2023-11-29:
> In PY32F030 datasheet, there is a table of sleep mode current, these numbers should work for all 002A/003/030. ![Screenshot from 2023-11-28 00-29-34](https://private-user-images.githubusercontent.com/65589019/285941710-9ea07dfd-50e8-42fb-aa04-94e0de20091d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDEyMjczNDMsIm5iZiI6MTcwMTIyNzA0MywicGF0aCI6Ii82NTU4OTAxOS8yODU5NDE3MTAtOWVhMDdkZmQtNTBlOC00MmZiLWFhMDQtOTRlMGRlMjAwOTFkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzExMjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMTI5VDAzMDQwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRlOGM5YWM3YjY2MTRmNDY3OGMyNjJmNzMxZmMyODI4YWFiODIwZTlkNGEyMjEzNmI0OGY4Zjg0NjFlNWQ5ZjYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.XufRSWBnmJPPvvj8HIEXh2ec7q7Xiiu-CGXVFJ3B9EY).

Thanks. So I think its deep sleep current is very high. However, its working voltage range is very good

IOsetting, 2023-11-29:
What I post is not **deep** sleep. Deep sleep is the stop mode, of which you can find more information [here](https://silicognition.com/Electronics/PuyaCharge/py32f003-undervoltage-stop1-mode/). Stop mode + LPTIM wakeup is a good solution for battery power saving.

garudaonekh, 2023-11-29:
4.7ua is good. Thanks. I will test it when I received the chip

garudaonekh, 2024-01-26:
HI,
Does your current implementation support Stop Mode and Sleep mode?

kimstik, 2024-01-26:
Please also take a look at the BIAS_CR register.

garudaonekh, 2024-01-26:
> Please also take a look at the BIAS_CR register.

Thanks. I take a look at it but it's too overwhelm complicated for me at this low level register code.

IOsetting, 2024-01-27:
> HI, Does your current implementation support Stop Mode and Sleep mode?

You can use `LL_LPM_EnableSleep()` and  `LL_LPM_EnableDeepSleep()` to enter Sleep mode and Stop mode. These are the built-in methods in [Puya's LL lib](https://github.com/IOsetting/py32f0-template/blob/main/Libraries/PY32F0xx_LL_Driver/Inc/py32f0xx_ll_cortex.h#L176)

garudaonekh, 2024-01-27:
> You can use `LL_LPM_EnableSleep()` and `LL_LPM_EnableDeepSleep()` to enter Sleep mode and Stop mode. These are the built-in methods in [Puya's LL [...]

Many Thanks, I will try and measure its current consumption and feedback.

IOsetting, 2024-02-05:
What is your circuit and measure method? And what is the measurement result?
For testing power consumption, please use a dedicated breakout board with solely this chip mounted on it. This ensures a clean and precise measurement of power usage.

closed 2024-02-19 by garudaonekh