IOsetting/py32f0-template #35: app crashes during UART transmit (printf) on PY32F002AL15S6TU

albassam, 2023-12-29:
I compiled the code without anychanges (except changing the targert in make file) using arm-none-eabi-gcc.exe (GCC) 12.2.0 and loaded the bin file to PY32F002AL15S6TU.

When I comment out printf the led is flashing as expected, otherwise it just crashes and writes garbage to the UART pin. Seems like buffer overflow issue but I haven't got the time to debug it. 

Has anyone faced similar issues?

IOsetting, 2023-12-29:
1. PF2 and PA2 are on the same pin, if you want to use PA2 as UART_TX, you need to update Option Byte to switch PF2 from NRST to GPIO.
2. Because PF2 is NRST, if there is a reset button with a 100nF capacitor between PF2 and GND, you need to remove this capacitor, or set the baudrate below 9600, otherwise the UART signal will be mostly filtered.

albassam, 2023-12-30:
I changed the options byte from 0000beaa to 0x0000feaa and now it worked for any baud rate!
Thanks again!

closed 2023-12-30 by albassam