IOsetting/py32f0-template #36: Unlocking and erasing proteced flash on PY32 chips

kholia, 2024-01-26:
Hi,

I am experimenting with flash protection on PY32 chips with the following program:

```C
#include <string.h>
#include "main.h"
#include "py32f0xx_bsp_clock.h"
#include "py32f0xx_bsp_printf.h"

static void APP_GPIOConfig(void);

static void APP_FlashSetOptionBytes(void)
{
  FLASH_OBProgramInitTypeDef OBInitCfg;

  LL_FLASH_Unlock();
  LL_FLASH_OB_Unlock();

  OBInitCfg.OptionType = OPTIONBYTE_USER;
  OBInitCfg.USERType = OB_USER_BOR_EN | OB_USER_BOR_LEV | OB_USER_IWDG_SW | OB_USER_WWDG_SW | OB_USER_NRST_MODE | OB_USER_nBOOT1;
  OBInitCfg.USERConfig = OB_BOR_DISABLE | OB_BOR_LEVEL_3p1_3p2 | OB_IWDG_SW | OB_WWDG_SW | OB_RESET_MODE_RESET | OB_BOOT1_SYSTEM;
  LL_FLASH_OBProgram(&OBInitCfg);

  LL_FLASH_OB_RDP_LevelConfig(OB_RDP_LEVEL_1);

  LL_FLASH_Lock();
  LL_FLASH_OB_Lock();
  /* Reload option bytes */
  LL_FLASH_OB_Launch();
}

int main(void)
{
  /* Set clock = 8MHz */
  BSP_RCC_HSI_8MConfig();

  /* Enable peripheral clock */
  LL_IOP_GRP1_EnableClock(LL_IOP_GRP1_PERIPH_GPIOA | LL_IOP_GRP1_PERIPH_GPIOB);

  BSP_USART_Config(115200);
  printf("SPI Demo: nRF24L01 Wireless\r\nClock: %ld\r\n", SystemCoreClock);

  APP_GPIOConfig();

  LL_mDelay(7000);

  uint32_t rdplvl = READ_BIT(FLASH->OPTR, FLASH_OPTR_RDP);
  if (rdplvl == OB_RDP_LEVEL_0) {
    APP_FlashSetOptionBytes();
  }
  else
  {
    printf("RESET has been configurated as RESET\r\n");
  }

  while (1)
  {
    printf("nRF24L01 check: error\r\n");
    LL_mDelay(2000);
  }
}

static void APP_GPIOConfig(void)
{
  LL_GPIO_InitTypeDef GPIO_InitStruct;

  // PA6 CSN
  LL_GPIO_SetPinMode(GPIOA, LL_GPIO_PIN_6, LL_GPIO_MODE_OUTPUT);
  // PA5 CE
  LL_GPIO_SetPinMode(GPIOA, LL_GPIO_PIN_5, LL_GPIO_MODE_OUTPUT);
  /* PA4 as input */
  GPIO_InitStruct.Pin = LL_GPIO_PIN_4;
  GPIO_InitStruct.Mode = LL_GPIO_MODE_INPUT;
  GPIO_InitStruct.Pull = LL_GPIO_PULL_UP;
  LL_GPIO_Init(GPIOA, &GPIO_InitStruct);
}

void APP_ErrorHandler(void)
{
  while (1);
}

#ifdef  USE_FULL_ASSERT
void assert_failed(uint8_t *file, uint32_t line)
{
  while (1);
}
#endif /* USE_FULL_ASSERT */
```

After power-cycling the PY32 chip (FY32F003 in SOP-8 package), I can no longer erase the chip using J-Link.

Is there a known method to `Unlock` and `Erase` the protected  flash memory on PY32 chips?

https://dzone.com/articles/unlocking-and-erasing-flash is a good read on this topic but I didn't a method which works for PY32 chips yet.

Thanks for the help.

IOsetting, 2024-01-26:
In [David's post](https://www.eevblog.com/forum/microcontrollers/$0-11-py32f002a-m0-24mhz-320kb-actually-324kb-more-peripherals/msg4937602/#msg4937602) he mentioned 

> After setting RDP1, I could no longer program or erase it with J-Link, I had to use PUYA ISP, setting BOOT=1 to enter the serial bootloader and running "Clear chip".

I guess you are in the same situation, but the problem is that SOP8(and SOP16) package has no BOOT pin so you cannot use PuyaISP to reset chip. 

If JLink can still detect and connect to the chip, maybe you can use the init and exit steps to unlock and change the option bytes as described in [this post](https://forum.segger.com/index.php/Thread/3144-SOLVED-J-Link-and-STM32F405RG-option-byte/)

deividAlfa, 2024-02-28:
Instead using PuyaISP everytime, I added a erase function triggered by special sequence of keys.

```
static void rdp_unlock(void){
  FLASH_OBProgramInitTypeDef pOBInit;
  LL_FLASH_OBGetConfig(&pOBInit);
  pOBInit.OptionType = OPTIONBYTE_RDP | OPTIONBYTE_USER;
  pOBInit.USERConfig = OB_BOR_DISABLE | OB_BOR_LEVEL_3p1_3p2 | OB_IWDG_SW | OB_WWDG_SW | OB_RESET_MODE_RESET | OB_BOOT1_SYSTEM;      // Default settings
  pOBInit.RDPLevel = OB_RDP_LEVEL_0;

  LL_FLASH_Unlock();
  LL_FLASH_OB_Unlock();
  LL_FLASH_OBProgram(&pOBInit);
  LL_FLASH_OB_Launch();
  NVIC_SystemReset();    // Just in case
}
```

After thisthe MCU will be erased (Flash erase is forced by hardware when clearing RDP setting).
Then cycle the MCU power for RDP to completely unlock, you'll be able to program it again.

deividAlfa, 2024-03-02:
Also, I managed to hard-brick some after disabling the bootloader and a bug in the unlock sequence.
It still can be done, as in this mode BOOT0=1 boots into SRAM.

Modify the linker script so everything is loaded into RAM:
```
/*
******************************************************************************
**
**  File        : LinkerScript.ld
**
**  Abstract    : Linker script for PY32F030x6 series
**                Set heap size, stack size and stack location according
**                to application requirements.
**                Set memory bank area and size if external memory is used.
**
**  Distribution: The file is distributed “as is,” without any warranty
**                of any kind.
**
*****************************************************************************
*/

/* Entry Point */
ENTRY(Reset_Handler)

/* Highest address of the user mode stack */
_estack = ORIGIN(RAM) + LENGTH(RAM);    /* end of RAM */
/*
  Generate a link error if heap and stack don't fit into RAM.
  These numbers affect the USED size of RAM
*/
_Min_Heap_Size = 0x0;   /* required amount of heap: 512 bytes */
_Min_Stack_Size = 0x0;  /* required amount of stack: 1024 bytes */

/* Specify the memory areas */
MEMORY
{
  RAM (xrw)      : ORIGIN = 0x20000000, LENGTH = 4K
  FLASH (rx)     : ORIGIN = 0x08000000, LENGTH = 32K
}

/* Define output sections */
SECTIONS
{
  
 /* The startup code goes first into FLASH */
  .isr_vector :
  {
    . = ALIGN(4);
    KEEP(*(.isr_vector)) /* Startup code */
    . = ALIGN(4);
  } >RAM

  /* The program code and other data goes into FLASH */
  .text :
  {
    . = ALIGN(4);
    *(.text)           /* .text sections (code) */
    *(.text*)          /* .text* sections (code) */
    *(.glue_7)         /* glue arm to thumb code */
    *(.glue_7t)        /* glue thumb to arm code */
    *(.eh_frame)

    KEEP (*(.init))
    KEEP (*(.fini))

    . = ALIGN(4);
    _etext = .;        /* define a global symbols at end of code */
  } >RAM

  /* Constant data goes into FLASH */
  .rodata :
  {
    . = ALIGN(4);
    *(.rodata)         /* .rodata sections (constants, strings, etc.) */
    *(.rodata*)        /* .rodata* sections (constants, strings, etc.) */
    . = ALIGN(4);
  } >RAM

  .ARM.extab   : { *(.ARM.extab* .gnu.linkonce.armextab.*) } >FLASH
  .ARM : {
    __exidx_start = .;
    *(.ARM.exidx*)
    __exidx_end = .;
  } >RAM

  .preinit_array     :
  {
    PROVIDE_HIDDEN (__preinit_array_start = .);
    KEEP (*(.preinit_array*))
    PROVIDE_HIDDEN (__preinit_array_end = .);
  } >RAM
  .init_array :
  {
    PROVIDE_HIDDEN (__init_array_start = .);
    KEEP (*(SORT(.init_array.*)))
    KEEP (*(.init_array*))
    PROVIDE_HIDDEN (__init_array_end = .);
  } >RAM
  .fini_array :
  {
    PROVIDE_HIDDEN (__fini_array_start = .);
    KEEP (*(SORT(.fini_array.*)))
    KEEP (*(.fini_array*))
    PROVIDE_HIDDEN (__fini_array_end = .);
  } >RAM

  /* used by the startup to initialize data */
  _sidata = LOADADDR(.data);

  /* Initialized data sections goes into RAM, load LMA copy after code */
  .data : 
  {
    . = ALIGN(4);
    _sdata = .;        /* create a global symbol at data start */
    *(.data)           /* .data sections */
    *(.data*)          /* .data* sections */
    *(.RamFunc)        /* .RamFunc sections */
    *(.RamFunc*)       /* .RamFunc* sections */
    
    . = ALIGN(4);
    _edata = .;        /* define a global symbol at data end */
  } >RAM AT> RAM

  
  /* Uninitialized data section */
  . = ALIGN(4);
  .bss :
  {
    /* This is used by the startup in order to initialize the .bss secion */
    _sbss = .;         /* define a global symbol at bss start */
    __bss_start__ = _sbss;
    *(.bss)
    *(.bss*)
    *(COMMON)

    . = ALIGN(4);
    _ebss = .;         /* define a global symbol at bss end */
    __bss_end__ = _ebss;
  } >RAM

  /* User_heap_stack section, used to check that there is enough RAM left */
  ._user_heap_stack :
  {
    . = ALIGN(8);
    PROVIDE ( end = . );
    PROVIDE ( _end = . );
    . = . + _Min_Heap_Size;
    . = . + _Min_Stack_Size;
    . = ALIGN(8);
  } >RAM

  /* Remove information from the standard libraries */
  /DISCARD/ :
  {
    libc.a ( * )
    libm.a ( * )
    libgcc.a ( * )
  }

  .ARM.attributes 0 : { *(.ARM.attributes) }
}
```