IOsetting/py32f0-template #37: ADC support on PY32F002A

Kanken6174, 2024-02-28:
Hi, i'm currently trying to get the LL ADC demos to work on the py32F002A, the basic led blink/uart demos work just fine but the ADC demo makes use of DMA features which don't exist on this specific MCU (030 and 003 only), even after commenting out the dma related lines the demo doesn't seem to work (no uart output besides the first line stating the name of the demo and the clock speed).

Do you have any idea what would need to be changed to acquire ADC data on the 002A?

IOsetting, 2024-02-28:
Just pretending it is a PY32F030, in Makefile change the type to PY32F030x6
```
# MCU types: 
#   PY32F002Ax5
#   PY32F002Bx5
#   PY32F003x6, PY32F003x8, 
#   PY32F030x6, PY32F030x8, 
#   PY32F072xB
MCU_TYPE		= PY32F030x6
```

Kanken6174, 2024-02-28:
And... that worked! I have honestly *no* idea how dma-enabled code can work without the hardware being present but it just does? Thank you.

closed 2024-02-28 by Kanken6174

prosper00, 2024-02-28:
> And... that worked! I have honestly _no_ idea how dma-enabled code can work without the hardware being present but it just does? Thank you.

It's because the 002A *is* an 030 internally. Same die, same hardware. The 030 features are not guaranteed to work - possibly the dies are defective in some way and that's why theyre binned down. But in all likelihood, they're perfectly functional and binned down for market segregation reasons, not technical ones. Also it's possible (but unlikely) that the 002 will get its own die down the road without these features. More likely is that the 002A will be discontinued or NRND'd, and replaced by a proper 'reduced' version (like the 002B) that doesn't have any of these hidden features.

In any case, it's fine for hobby stuff, but I'd be hesitant to design a commercial product that depends on DMA and 030 features on an 002A.

Kanken6174, 2024-02-28:
That's a very interesting fact i didn't know about, i guess i'll just enjoy the free DMA hardware for now then 😄
thanks for the heads-up