IOsetting/py32f0-template #38: Help with LSI  as Clock Source to reduce power consumption

garudaonekh, 2024-03-26:
Hi, 
As per Py32f030 datasheet, with LSI as clock source the power consumption in run mode is around 0.1ma.

I set the Low Speed Clock Source to LSI 32.768KHZ but it's not working. The power consumption still at around 0.4ma.

I try to set it in HAL mode also but still not working

```
RCC_OscInitTypeDef OSCINIT;
  RCC_PeriphCLKInitTypeDef LPTIM_RCC;

  /* LSI时钟配置 */
  OSCINIT.LSEState=RCC_LSE_OFF;
  OSCINIT.HSIState=RCC_HSI_OFF;
  OSCINIT.OscillatorType = RCC_OSCILLATORTYPE_LSI;  /* 选择配置LSI */
  OSCINIT.LSIState = RCC_LSI_ON;                    /* LSI开启 */
```
Any configuration I am missing?

Thanks;

garudaonekh, 2024-03-27:
I tried to use the OpenPuya LL example, but setting LSI as clock source hang the MCU and stuck there.
```

static void APP_SystemClockConfig(void)
{
  
  /* LSI使能及初始化 */
  LL_RCC_LSI_Enable();
  while(LL_RCC_LSI_IsReady() != 1)
  {
  }
  
  /* 设置AHB分频 */
  LL_RCC_SetAHBPrescaler(LL_RCC_SYSCLK_DIV_1);
  
  /* 配置LSI为系统时钟及初始化 */
  LL_RCC_SetSysClkSource(LL_RCC_SYS_CLKSOURCE_LSI);
  while(LL_RCC_GetSysClkSource() != LL_RCC_SYS_CLKSOURCE_STATUS_LSI)
  {
  }
  
  LL_FLASH_SetLatency(LL_FLASH_LATENCY_0);
  
  /* 设置APB1分频及初始化 */
  LL_RCC_SetAPB1Prescaler(LL_RCC_APB1_DIV_1);
  /* 更新系统时钟全局变量SystemCoreClock(也可以通过调用SystemCoreClockUpdate函数更新) */
  LL_SetSystemCoreClock(LSI_VALUE);
}
```

https://github.com/OpenPuya/PY32F0xx_Firmware/blob/master/Projects/PY32F030-STK/Example_LL/RCC/RCC_LSI_OUTPUT/Src/main.c

IOsetting, 2024-04-30:
Apologies, I had read the question for a while but I didn't know what to answer. 
I didn't get time and parts to solder a clean board to test these power consumption numbers. 
Recently I have got what I need and did some tests. I guess you need to enable Flash Sleep mode in LSI to make 100uA, please find the example code in above commit.

garudaonekh, 2024-05-01:
thanks for taking time to make a specific example on this. I will try it

garudaonekh, 2024-05-04:
It works. Many thanks

closed 2024-05-04 by garudaonekh