IOsetting/py32f0-template #40: [question] What is the story of this repository?

cr1tbit, 2024-04-13:
I want to understand where these examples (actually a lot of them!) come from. They look a bit like generated from a codegen tool similar to stm32cubemx. 

I was trying to find if puya provides such tool, but it's hard to google this on the western side of the internet.

The actual problem I'd like to solve is running the `HAL_I2C_MODE_MEM` mode, in hopes of emulating a generic IO expander as close as possible. This would be extremely useful, seeing that any I2C ADC or IO expander are more expensive than this MCU, let alone anything with more fancy utils like charlieplexing, built-in conditions, etc.

IOsetting, 2024-04-13:
Puya doesn't provide such tool as stm32cubemx, most of the examples were modified based on Puya's SDK, which can be found at https://github.com/OpenPuya/PY32F0xx_Firmware/tree/master/Projects/

cr1tbit, 2024-05-26:
Thanks for explanation :)

For anyone reading this issue in the future, I created a PR with goal of implementing the aforementioned features. https://github.com/IOsetting/py32f0-template/pull/48

closed 2024-05-26 by cr1tbit