IOsetting/py32f0-template #41: Examples of I2C DMA

jonnor, 2024-04-20:
Hi,
first of all: thank you for this fantastic repository. The build system setup is super pragmatic, easy to work with - and everything I have tried so far works nicely. And there is an impressive amount of examples.

I have one project with an accelerometer, where I would like to read at high rates (from the built-in FIFO). And it is also doing computationally intensive processing (some DSP and machine learning). So I would be very interested in using the DMA support to enable doing work on the CPU while the transfers are ongoing. I see that there is a `DMA_CHANNEL_MAP_I2C*` in the code. So there does seem to be support for this. There are DMA examples for the ADC for example, but I have not seen anything for I2C+DMA. Does anyone have such examples and/or tips&tricks? It is basically just receive that would benefit from DMA. SPI+DMA would also be relevant

IOsetting, 2024-04-20:
I haven't tried I2C DMA, in my work I use timer to trigger I2C reading. 
Have you tried the I2C DMA examples in Puya's SDK? https://github.com/OpenPuya/PY32F0xx_Firmware/tree/master/Projects/PY32F030-STK/Example_LL/I2C

deividAlfa, 2024-07-02:
An example sending 1KB buffer to an I2C OLED display using DMA:

```
/*
 * display.c
 *
 *  Created on: 31 dic. 2022
 *      Author: David
 */

#include "display.h"
#include "delay.h"

#if defined (ENABLE_OLED_SPI) || defined ( ENABLE_OLED_I2C)

u8g2_t u8g2;
volatile uint8_t oled_busy;

#ifdef OLED_128X32
  #define OLED_ROWS 4
#elif defined OLED_128X64
  #define OLED_ROWS 8
#endif

__attribute__((aligned(4)))uint8_t oled_buffer[128*OLED_ROWS];

ErrorStatus display_present;

const uint8_t disp_init[] = {             // Initialization for SH1106 / SSD1306 / SSD1309

    1, c_disp_off,
    2, c_clock_set, 0xF0,                 // Set max framerate

#ifdef OLED_128X32
    2, c_mux_ratio, 0x1F,           // 128x32
    1, c_remap_off,
    1, c_com_norm,
    2, c_com_cfg,   0x02,     // 128x32

#elif defined OLED_128X64
    2, c_mux_ratio, 0x3F,         // 128x64
    1, c_remap_on,
    1, c_com_rev,
    2, c_com_cfg,   0x12,   // 128x64
#else
    #error Display type not defined at display.h !
#endif

    2, c_offset,    0x00,
    1, c_start_line,
    2, c_addr_mode, 0x02,

    2, c_contrast,  0xfF,                 // Init in max contrast
    2, c_precharge, 0xff,                 // Set Pre-Charge Period, 0x44 (4 Display Clocks [Phase 2] / 4 Display Clocks [Phase 1])
    2, c_vcomh_lvl, 0xff,                 // Set VCOMH Deselect Level, 0x3C (0.84*VCC)
    1, c_all_off,
    1, c_inv_off,
    2, c_pump_1306_set, c_pump_1306_on,   // For SSD1306
    2, c_pump_1106_set, c_pump_1106_on,   // For SH1106
    1, c_pump_1106_adj | 0x03             // For SH1106, pump to 9V
};

#if defined (ENABLE_OLED_I2C)

ErrorStatus I2C1_MasterTransmit(uint8_t DevAddress, uint8_t mode, uint8_t *pData, uint16_t Size)
{
  ErrorStatus status = SUCCESS;
  LL_I2C_GenerateStartCondition(I2C1);
  while(LL_I2C_IsActiveFlag_SB(I2C1) != 1);

[... 221 lines cut ...]
  LL_DMA_SetDataLength(DMA1, LL_DMA_CHANNEL_1,128);
  LL_DMA_EnableChannel(DMA1, LL_DMA_CHANNEL_1);

  LL_I2C_GenerateStartCondition(I2C1);
  while(!LL_I2C_IsActiveFlag_SB(I2C1));

  LL_I2C_TransmitData8(I2C1, OLED_ADDR);
  while(!LL_I2C_IsActiveFlag_ADDR(I2C1));
  LL_I2C_ClearFlag_ADDR(I2C1);

  LL_I2C_TransmitData8(I2C1, modeData);
  while(!LL_I2C_IsActiveFlag_TXE(I2C1));

  LL_I2C_EnableDMAReq_TX(I2C1);
  row++;
}
#endif
```

## py32f0xx_it.c
```
void DMA1_Channel1_IRQHandler(void){
  display_isr();
}
```