IOsetting/py32f0-template #42: Current Consumption in Deep Sleep LSI

garudaonekh, 2024-05-03:
In this example:
https://github.com/IOsetting/py32f0-template/blob/main/Examples/PY32F002B/LL/LPTIM/LPTIM_LSI_DeepStop_WakeUp/main.c

Its description said it consumes **around 0.6uA in deep stop mode.**

Is this figure real? As per its datasheet, it should be around 5ua or above.

I don't have Py32F002B to test, I have only F002A.

IOsetting, 2024-05-03:
002B is a different series, low power but less on-chip resources.

It's real but not officially guaranteed. In 002B's datasheet it only lists the stop mode. Deep stop mode is a feature of PY32L020, but I doubt they are the same chip.

IOsetting, 2024-05-03:
All datasheets can be found at [WIKI](https://github.com/IOsetting/py32f0-template/wiki) page

garudaonekh, 2024-05-03:
[image]
Glad to know that PY32f002B is much more power efficient than others

closed 2024-05-06 by garudaonekh