IOsetting/py32f0-template #45: Microsecond delay

garudaonekh, 2024-05-09:
Hi,
I need to migrate code from Arduino to LL which use **delayMicroseconds()**, which is not availabe in LL as far as I know.

I come across this code of microseconds delay, is it correct?
```

 uint32_t   wait_loop_index = (usDelay * (SystemCoreClock / 1000000U));
                while(wait_loop_index != 0U)
                {
                    wait_loop_index--;
                }
```

Thanks;

garudaonekh, 2024-05-09:
I came across a function in OpenPuya which is

 ```
void APP_uDelay(uint32_t nus)
{
  __IO uint32_t Delay =1+ nus * (SystemCoreClock / 24U) / 1000000U;
  do
  {
    __NOP();
  }
  while(Delay--);
}
```

Sorry, I am from Arduino background so this tickcount and clock is too overwhelmed for me.

IOsetting, 2024-05-09:
Considering the flash delay, I guess this one is closer.

> ```
> void APP_uDelay(uint32_t nus)
> {
>  __IO uint32_t Delay =1+ nus * (SystemCoreClock / 24U) / 1000000U;
>  do
>  {
>    __NOP();
>  }
>  while(Delay--);
> }
> ```

garudaonekh, 2024-05-09:
> Considering the flash delay, I guess this one is closer.
> 
> > ```
> > void APP_uDelay(uint32_t nus)
> > {
> >  __IO uint32_t Delay =1+ nus * (SystemCoreClock / 24U) / 1000000U;
> >  do
> >  {
> >    __NOP();
> >  }
> >  while(Delay--);
> > }
> > ```

I do not think so. I try to check the printf timestamps. The closest is  / 4U. But I need more accurate one for using in RF433 signalling

garudaonekh, 2024-05-16:
Convert this code from Arduino and it works for my case.
```
void delayMicroseconds2(uint32_t us)
{
  __IO uint32_t currentTicks = SysTick->VAL;
  /* Number of ticks per millisecond */
  uint32_t tickPerMs = (SysTick->LOAD + 1)/1000;
  /* Number of ticks to count */
   uint32_t nbTicks = ((us - ((us > 0) ? 1 : 0)) * tickPerMs);
  /* Number of elapsed ticks */
  tickPerMs=tickPerMs*1000;
  uint32_t elapsedTicks = 0;
  __IO uint32_t oldTicks = currentTicks;
  do {
    currentTicks = SysTick->VAL;
    elapsedTicks += (oldTicks < currentTicks) ? tickPerMs + oldTicks - currentTicks :
                    oldTicks - currentTicks;
    oldTicks = currentTicks;
  } while (nbTicks > elapsedTicks);

}
```

closed 2024-05-16 by garudaonekh