IOsetting/py32f0-template #46: ADC_COMP_IRQn on PY32F002A

axel-angel, 2024-05-16:
Hello,

Thank you a lot for this repository, it's very useful!
I found a possible confusing issue: I'm using ADC on 02A with ADC_COMP_IRQn interrupt and I found out that it never triggers if I compile with `MCU_TYPE = PY32F002Ax5` but it works fine when compiling for PY32F003x6 . The code is `Examples/PY32F0xx/LL/ADC/ADC_SingleConversion_TriggerTimer` . 

Here is what I tried and found: (a) I tried my own code as well, same issue. (b) I also tried without interrupt: busy wait like example `ADC_InternalTemperature` this works fine in all cases so ADC is working properly. (c) I don't know how to debug with SWD and OCD yet so I used a led to indicate if interrupt is called. (d) my project is using other interrupts like TIM16_IRQn, EXTI2_3_IRQn, etc so I know other interrupts work, the code to enable them too.

I came to conclusion that the ADC is working but the interrupt ADC_COMP_IRQn is either (1) never enabled properly if `MCU_TYPE = PY32F002Ax5`  when doing this:
```C
  LL_ADC_EnableIT_EOC(ADC1); // interrupt
  NVIC_SetPriority(ADC_COMP_IRQn, 0);
  NVIC_EnableIRQ(ADC_COMP_IRQn);
  LL_ADC_REG_StartConversion(ADC1); // start conversion, should trigger interrupt
```
or (2) the interrupt is raised but its routine is never called through IRQ. I'm out of idea how to debug this. I thought to report as it may be a bug, and very sneaky one. I have checked the headers and similar device-specific files, I couldn't find an obvious difference about the ADC.

IOsetting, 2024-05-16:
In Libraries/CMSIS/Device/PY32F0xx/Source/gcc/startup_py32f002a.s, the ADC exception handler is *ADC_IRQHandler*, 
```asm
  .word  ADC_IRQHandler                 /* 12ADC */
...
  .weak      ADC_IRQHandler                
  .thumb_set ADC_IRQHandler,Default_Handler
```
change the name to *ADC_COMP_IRQHandler* might solve your problem

axel-angel, 2024-05-17:
Oh I see, that explains. Thank you!

closed 2024-05-17 by axel-angel