IOsetting/py32f0-template #47: Suppport for Py32L020

garudaonekh, 2024-05-18:
https://github.com/OpenPuya/PY32L020_Firmware

Hi, please add support for this new low power series

IOsetting, 2024-05-18:
I haven't bought PY32L020, but by comparing the register definitions, I would say F002B and L020 are the same chip.
Setting `MCU_TYPE = PY32F002Bx5` will work on PY32L020

garudaonekh, 2024-05-19:
Thanks, I will try it when I received the sample. Somehow, it has DEEP STOP mode[ https://github.com/OpenPuya/PY32L020_Firmware/tree/master/Projects/PY32L020-STK/Example/PWR/PWR_DEEPSTOP_WFI ], which is **0.7ua**

IOsetting, 2024-05-19:
PY32F002B also supports DEEP STOP in my test, the testing code is at [Examples/PY32F002B/LL/PWR/PWR_DeepStop](https://github.com/IOsetting/py32f0-template/tree/main/Examples/PY32F002B/LL/PWR/PWR_DeepStop)

closed 2025-01-30 by garudaonekh