IOsetting/py32f0-template #58: **Title**: `HAL_Delay()` Over-Delay Due to Incorrect `wait` Adjustment

jjsch-dev, 2025-03-24:
**Title**: `HAL_Delay()` Over-Delay Due to Incorrect `wait` Adjustment  

### **Description**  
The current implementation of `HAL_Delay()` adds `uwTickFreq` to the `wait` variable when `wait < HAL_MAX_DELAY` (where `HAL_MAX_DELAY = 0xFFFFFFFF`). For any delay value less than `0xFFFFFFFF`, this introduces an **unintended 1 ms over-delay** (assuming `uwTickFreq = 1`). This breaks timing-critical applications (e.g., GPIO toggling, PWM generation) and leads to inaccurate results.  

---

### **Observed Behavior**  
Calling `HAL_Delay(10)` in a loop to toggle a GPIO pin results in a **45.5 Hz signal** instead of the expected **50 Hz**.  

---

### **Root Cause**  
- The condition `wait < HAL_MAX_DELAY` is **always true** for practical delays (e.g., `10 < 0xFFFFFFFF`), forcing `wait += uwTickFreq` for all delays.  
- Example: `HAL_Delay(10)` → `wait = 10 + 1 = 11 ms` → Total period = 22 ms → **45.45 Hz**.  

---

### **Proposed Fix**  
Change the comparison to `wait < 1` to ensure `uwTickFreq` is **only added for `Delay = 0`**, preserving accuracy for `Delay > 0`:  
```c  
// Current code (problematic):  
if (wait < HAL_MAX_DELAY) // Always true for reasonable delays  
{  
  wait += uwTickFreq;     // Adds 1 ms even for non-zero delays  
}  

// Proposed fix:  
if (wait < 1)             // Only true for Delay = 0  
{  
  wait += uwTickFreq;     // Ensures HAL_Delay(0) waits at least 1 ms  
}  
```
---

### **Why This Works**
- **Overflow Handling:** The loop logic (` HAL_GetTick() - tickstart) < wait`  inherently handles 32-bit counter overflows via unsigned integer wrapping. No additional checks are needed.

- **Intent Preservation:** The original comment (` /* Add a freq to guarantee minimum wait */` ) suggests the adjustment was meant for ` Delay = 0`  to enforce a minimal delay. This fix aligns with that intent.

---
### **Impact:**
- **For Delay > 0:** Timing becomes accurate (` e.g., HAL_Delay(10) → 10 ms` ).

- **For Delay = 0:** A minimal delay of 1 ms is enforced (if required).

### **Steps to Reproduce:**
- Use this code to toggle a GPIO pin:

```c  
while (1)  
{  
  HAL_GPIO_TogglePin(LED_GPIO_PORT, LED_GPIO_PIN);  
  HAL_Delay(10);  
}  
``` 
- Measure the output frequency with an oscilloscope.

- **Observed:** ~45.5 Hz. **Expected:** 50 Hz.

[image]

---
### **Request**
- Update the **HAL_Delay()** logic to resolve this timing inaccuracy.
- Monitor [OpenPuya#2](https://github.com/OpenPuya/PY32F0xx_Firmware/issues/2) for updates.
---
### **Additional Context**
- Similar issues were resolved in STM32’s HAL by removing the extra tick addition ([STM32 Forum Thread](https://community.st.com/s/question/0D53W00001EJh6pSAD/hal-delay-adds-1ms-extra)).

- Testing confirms that changing the condition to `wait < 1` restores the expected 50 Hz signal.

[image]

Thank you!

IOsetting, 2025-04-01:
Thank you for your detailed explanation, I will create a fix for this.

jjsch-dev, 2025-04-02:
Thank you! Let me know if you need help testing the fix. Happy to assist! 😊

Best regards,
Juan

jjsch-dev, 2025-05-04:
@IOsetting, thanks for implementing the solution I suggested to resolve this issue. I tested it and it works perfectly!

IOsetting, 2025-05-09:
Thank you! I will close this.

closed 2025-05-09 by IOsetting