IOsetting/py32f0-template #60: [not an issue] Re-purposing a Baofeng BF-666S board with a PY32F002AF15P6TU

cocus, 2025-08-26:
Hello all,
First of all, this is an awesome project! I've done something similar for another cheap chinese chip (PHY6222) so I know the hurdles of navigating through poorly supported SDKs and chips.
In any case, I have multiple boards of walkie talkies from Baofeng, which have one of the PY32F002AF15P6TU, and an integrated walkie talkie chip. No documentation on said chip is available online (AT1141). I've traced the footprint to be similar to the RDA1846 (at least the power lines, i2c lines, etc).
However, doing a quick I2C logic dump reveals that it's indeed NOT a RDA1846 (even if it uses the same address!). In any case, I'm tracing the connections from all the chips on the board to the Puya MCU; I already have maped that the I2C lines are connected to PA0 and PA1 (which is weird since this MCU doesn't have I2C as an alternative functionality of these pads, but whatever).
There are the SWD pins exposed on the backside of the board, but I've figured that either the MCU re-purposes those pins for something else, and/or the NRST is also used for something different than reset. I've verified that pulling NRST low and powering up the board makes it to not do absolutely anything; but once it "starts", pulling NRST low doesn't reset or holds the processor in reset.
In any case, I'm using a J-Link ARM probe, I've tried the J-Flash with the JLinkDevices + jlink-script, and openocd with a custom config file (stolen from the stm32f0x, which I don't know which DAP tap id to use, so that's mostly what I wanted to ask here your thoughts); but nothing works. There's no reply from the device, and I suspect it's because they're using the SWDIO and SWDCLK pins for something else. Even if I connect the NRST pin to the J-Link probe, it can't reset it (as I've mentioned above).

In any case, the cfg file for openocd contains this (even if it's not working):
```
# SPDX-License-Identifier: GPL-2.0-or-later

# script for py32f02 family

#
# stm32 devices support SWD transports only.
#
source [find target/swj-dp.tcl]
source [find mem_helper.tcl]

if { [info exists CHIPNAME] } {
   set _CHIPNAME $CHIPNAME
} else {
   set _CHIPNAME py32f02
}

set _ENDIAN little

# Work-area is a space in RAM used for flash programming
if { [info exists WORKAREASIZE] } {
   set _WORKAREASIZE $WORKAREASIZE
} else {
   set _WORKAREASIZE 0xC00
}

# Allow overriding the Flash bank size
if { [info exists FLASH_SIZE] } {
	set _FLASH_SIZE $FLASH_SIZE
} else {
	# autodetect size
	set _FLASH_SIZE 0x5000
}

#jtag scan chain
if { [info exists CPUTAPID] } {
   set _CPUTAPID $CPUTAPID
} else {
   set _CPUTAPID 0x0bb11477 # this is not right
}

swj_newdap $_CHIPNAME cpu -irlen 4 -ircapture 0x1 -irmask 0xf -expected-id $_CPUTAPID
dap create $_CHIPNAME.dap -chain-position $_CHIPNAME.cpu

set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME cortex_m -endian $_ENDIAN -dap $_CHIPNAME.dap

$_TARGETNAME configure -work-area-phys 0x20000000 -work-area-size $_WORKAREASIZE -work-area-backup 0

# flash size will be probed
set _FLASHNAME $_CHIPNAME.flash
#######flash bank $_FLASHNAME stm32f1x 0x08000000 $_FLASH_SIZE 0 0 $_TARGETNAME

# adapter speed should be <= F_CPU/6. F_CPU after reset is 8MHz, so use F_JTAG = 1MHz
adapter speed 1000

adapter srst delay 100

reset_config srst_nogate

if {![using_hla]} {
   # if srst is not fitted use SYSRESETREQ to
   # perform a soft reset
   cortex_m reset_config sysresetreq
}

# Default hooks
#######$_TARGETNAME configure -event examine-end { stm32f0x_default_examine_end }
#######$_TARGETNAME configure -event reset-start { stm32f0x_default_reset_start }
#######$_TARGETNAME configure -event reset-init { stm32f0x_default_reset_init }

```

I'm NOT changing the BOOT0 pin at all, it's held with its default value from the original circuit (which I think it's pulling it low? can't trace where it's connected though)

I'd appreciate any feedback you can give!

Thanks.

cocus, 2025-08-29:
Ok, getting back to this. I've seen some weird references to functions and data beyond the 20K of the flash, like the following code:

[image]
or

[image]

Is there an undocumented ROM after the flash? or is it that the flash is bigger? I'm pretty sure this is a 20K part, so this is a little bit puzzling. I'm still not sure if I've dumped the actual firmware or if this is the bootloader? I don't think so... but it has those references to odd memory locations, and it doesn't make any sense.

Well, in any case, any feedback is welcome!

Thanks!

cocus, 2025-08-29:
I've re-connected the target, and dumped an additional 12kb out of it, so the entire ROM is 32kB! the contents are fine, and OpenOCD doesn't complain when reading that section in between 0x08005000 - 0x08008000 which is not marked as anything on the datasheet!

[image]

However, if I try to read from 0x08008000 or something in that "reserved" space, I do get errors on OpenOCD, which is expected. I'm thinking that *MAYBE* this thing has a larger flash than advertised?

IOsetting, 2025-09-04:
> However, if I try to read from 0x08008000 or something in that "reserved" space, I do get errors on OpenOCD, which is expected. I'm thinking that _MAYBE_ this thing has a larger flash than advertised?

Yes, it's known that 002A has 32KB flash, [here](https://www.eevblog.com/forum/microcontrollers/$0-11-py32f002a-m0-24mhz-320kb-actually-324kb-more-peripherals/) is the discussion.

My apologies I cannot help the rom/flash reading for I am not an expert in this field. I searched the keyword AT1141 but the information is very limited:

* There is a [copyright application](https://banquan.tianyancha.com/rj/em4ov99o6d32202696213313ald2v238) ([certificats list](http://www.qzcxdz.cn/product/16/)) by company named 泉州市创新电子科技有限公司(Quanzhou Chuangxin Electronic Technology Co., Ltd) in 2022 for their software *New wireless intercom technology program software based on AT1141*, Baofeng company is also in Quanzhou so they might have business connections
* RF chip AT1141 is a production of [AUCTUS](https://en.auctus.cn/) but they doesn't provide datasheet downloads, and there is no information on the Internet. I'll contact them and see if I can get some technical information.

cocus, 2025-09-04:
> Yes, it's known that 002A has 32KB flash, [here](https://www.eevblog.com/forum/microcontrollers/$0-11-py32f002a-m0-24mhz-320kb-actually-324kb-more-peripherals/) is the discussion.

Excellent, I tried to find something about it but... not much. that's good!

> My apologies I cannot help the rom/flash reading for I am not an expert in this field. I searched the keyword AT1141 but the information is very limited:

Yes, I know this is kinda all private chips and such, thus why I'm reverse-engineering the firmware I've dumped from the puya. At first, most of it didn't make sense because I was missing the other 12KB of it (the firmware uses the 32kb!). Now I'm on track of figuring out what's going on. Nothing weird so far.

What I'd like to review is adding support for this chip on openocd. Once I feel I have mapped out the majority of the pins of the puya on this board, I'll try to make a simple blinky project and flash it through jlink, and start tinkering with flash routines for openocd. I'll try to commit these and open a PR. 

This project is really cool and I've used it to map out the peripherals that the firmware is using, through the .h files and such. Thanks again for it.

I'll update this issue once I have more information.